OrcaSlicer/OrcaSlicer #9537: Retraction Speed is in gcode the distance too

juckendesAuge, 2025-05-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer_Linux_AppImage_Ubuntu2404_V2.3.0.AppImage

### Operating System (OS)

Linux

### OS Version

Fedora 42

### Printer

Prusa MK3S.5

### How to reproduce

The retraction speed is also used for the distance. The 0.8mm retraction distance is not displayed at all in the gcode.

### Actual results

I don't have access to my PC at the moment. But I wanted to report the bug anyway.

### Expected results

I don't have access to my PC at the moment. But I wanted to report the bug anyway.

### Project file & Debug log uploads

I don't have access to my PC at the moment. But I wanted to report the bug anyway.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MxBrnr, 2025-05-04:
Please upload the sliced G-code for this when you get the chance.

juckendesAuge, 2025-05-04:
> Please upload the sliced G-code for this when you get the chance.

[image]
[image]
[image]

igiannakas, 2025-05-08:
Have you checked the filament profile in case there is a settings override for the retraction distance?

juckendesAuge, 2025-05-08:
Thank you very much 

Yes, I noticed that last night, plus the 70% retraction in the printer itself under certain circumstances. With the filament overrides, you then have to calculate 70% to get the gcode values.
This means that Orca Slicer works as it should.

As I am still on the road, I wanted to write all this on the PC first. 😅

I apologize for the inconvenience.

closed 2025-08-15 by github-actions[bot]