OrcaSlicer/OrcaSlicer #9539: "Check For Updates" does nothing.

B-Man1, 2025-05-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

V2.2.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Voron 2.4, K1 Max etc

### How to reproduce

"Check For Updates does nothing.  I am on V2.2.0.  I know V2.3.0 exists, yet "Check For updates" does not say it does or does not.  Instead it does literally nothing

### Actual results

Nothing

### Expected results

Update

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kisslorand, 2025-05-08:
Update your OrcaSlicer.

https://github.com/SoftFever/OrcaSlicer/releases

https://orcaslicer.com

B-Man1, 2025-05-08:
> Update your OrcaSlicer.
> 
> https://github.com/SoftFever/OrcaSlicer/releases
> 
> https://orcaslicer.com

I did thanks

closed 2025-08-15 by github-actions[bot]