OrcaSlicer/OrcaSlicer #9540: 2.3.0: sense of printable check mark reversed

wrishel, 2025-05-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

windows 11 pro v 24H2 build 26100.3675

### Additional system information

Between 2.2 and 2.3 the sense of the check for printable in the context menu for an object has been reversed. It being checked use to mean the object would be printed. Now it means that it will not be printed. This is not intuitive. One would assume that the word being checked would mean the object is able to be printed.

### Printer

Bambu X1C

### How to reproduce

1. Right click on an object on the plate
2. Initially there will be no check mark next to the word printable and the object will be fully visible on the plate.
3. Click on printable. The check mark will appear and the object on the plate will show as a "ghost"

### Actual results

The object will no longer be sliced and printed.

### Expected results

As in 2.2 the item should be checked and fully visible. Clicking on printable should make the check mark disappear and the object on the plate become a ghost.

### Project file & Debug log uploads

[Bat Case #4 all three parts.zip](https://github.com/user-attachments/files/20026390/Bat.Case.4.all.three.parts.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This is pretty straightforward. I'm hoping this is enough

kisslorand, 2025-05-08:
It seems it is already fixed.

[image]

[image]

wrishel, 2025-05-08:
> It seems it is already fixed.
>
> C2CCACE6-219C-4162-8E55-97D9522284ED.png (view on web)
>
> 40DAF23E-4902-4A6A-B1E5-FC3BA0EDE6E2.png (view on web)

I wonder if you are on a nightly build or some version that is not the
official stable version. I certainly see the phenomenon on what I believe
to be the official version for ordinary people.

[image: image.png]

-- Wes

kisslorand, 2025-05-09:
Yes, I am on a nightly build. I was not debating whether the issue is present or not in the  latest release, I just stated that the issue has been  already taken care.

Noisyfox, 2025-05-09:
Dup #9059

closed 2025-05-09 by Noisyfox