OrcaSlicer/OrcaSlicer #9541: Dark Mode Adjustments

fsk-gjjb, 2025-05-04:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When usingf Dark Mode in Orca Slicer, the difference between the background and the build plate is very small, making it difficult to discern the edge of the build plate, especially on certain monitors.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would greatly increasy user experience if the background/buildplate could be set to different colours to increase distinction between them. As a short term fix could the current difference just be slightly adjusted with a full feature of being able to adjust it be implemented later.

### Describe alternatives you've considered

1. I have considered using Light mode but working all day behind the computer in light mode is strenious on the eyes.
2. Using another slicer is not an option. I like Orca and speaking for myself and possibly others, do not want to migrate.

tlhintoq, 2025-05-06:
This is what my dark mode looks like.  Not hard to discern the beds from the background.
Can you include a screen shot of yours?

[image]

[image]

tlhintoq, 2025-05-06:
Also... You could go in to the bitmap used for your printer and give it a border that works better for you... black or white lets say.  Just edit the file.  Like in my screen shot where it has "Elegoo" in white... give it a board around the plate like that in white.

CalvinSlaven, 2025-05-07:
how do i turn on dark mode?

fsk-gjjb, 2025-05-09:
Good day SgtSaint,

Thank you for your email and response to my request.

Attached are the snip in Windows 10 that I took as well as a photo of the
screen itself that I took with my phone. I have an Samsung 34 UltraWide
monitor (100Hz) and an nVidia RTX2060 (Everything on latest drivers, RTX on
Studio-driver).

I played around with the screen's settings for contrast etc, as well as the
nVidia app's brightness/contrast/hue etc settings but the 'issue'.with poor
contrast is more or less the same regardless of display/screen settings.
The issue also persists regardless of buildplate orientation or loaded
STL/model within Orca (latest version: 2.3.0, Build 70931e5). I was
actually surprised that your images have such well defined contrast, thanks
for attaching those!

Thank you once again for your response and time in this matter.

Kind regards,

Gert

fsk-gjjb, 2025-05-09:
Hi Sargeant,

Thanks for the tip about the bitmap edit.I watched a YT video quickly and
saw the 'Texture Import' that loads STL, is where this is done. I saw
the STL, but did not realise it can be used for defining bed texture as
well as just the colour/image! I will quickly whip up a model in Solidworks
that is lighter, or with a well defined border at least and import it.

If you wish, I think we can close this thread on GitHub since my issue can
be resolved with a very acceptable workaround.

Thank you once again for making the community work!

Kind regards,

Gert