OrcaSlicer/OrcaSlicer #9542: Better retraction calibration

Andreas-strg, 2025-05-04:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I think we need a beter retraction calibration to calibrate **length and speed** 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

There are some better methods on the Internet

http://retractioncalibration.com/ (https://www.youtube.com/watch?v=9mZnkIsibog)

MxBrnr, 2025-05-04:
Duplicate of https://github.com/SoftFever/OrcaSlicer/issues/6985. Please make sure you are searching for existing issues before posting a new one, especially since you indicated that you did.

Andreas-strg, 2025-05-04:
sorry, not found

closed 2025-05-04 by Andreas-strg