OrcaSlicer/OrcaSlicer #9552: Thumbnails not showing on Qidi X max 3

ipod9138, 2025-05-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

10 and 11

### Printer

Qidi X max 3

### How to reproduce

1. slice file
2. send file to printer
3. printer starts
4. no thumbnail of print on display screen just a black square

### Actual results

Just a black square where image of the file should be.

### Expected results

Should show an image of the file being printed on the display screen

### Project file & Debug log uploads

[CHEPCalibrationCube Orca Report.zip](https://github.com/user-attachments/files/20035990/CHEPCalibrationCube.Orca.Report.zip)

[debug_Mon_May_05_12_07_22_18984.log.zip](https://github.com/user-attachments/files/20035992/debug_Mon_May_05_12_07_22_18984.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

The drop down to select Colpic is missing, even though it says 380x380/Colpic, 210X210/Colpic, 110x110/PNG it still doesnt display the image
The drop down need to be re-added

A1B3F6, 2025-08-08:
I am using 2.3.0 and have the same issue. Gcode files created with version 2.0.0 the thumbnails work properly with my X-Max 3. It appears there is not much interest in fixing this as there are several unresolved tickets for this concern. Is it possible to go back to version 2.0.0?

ipod9138, 2025-08-08:
> I am using 2.3.0 and have the same issue. Gcode files created with version 2.0.0 the thumbnails work properly with my X-Max 3. It appears there is not much interest in fixing this as there are several unresolved tickets for this concern. Is it possible to go back to version 2.0.0?

I just rolled back to V2, look in GitHub orca slicer release history to find V2.

It’s a shame the devs can’t be ar$ed to sort it out.

ipod9138, 2026-11-07:
Installed Freedi OS/Ui to my Qidi Xmax 3's, it’s a better system all round and the issue was resolved.