OrcaSlicer/OrcaSlicer #9560: Internal bridges create extra solid infill line that causes print issues/failures

staal54a, 2025-05-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu X1C

### How to reproduce

1. Open OrcaSlicer and select printer (doesn't matter from my findings)
2. Add a primitive cube (Right click plate -> Add Primitive -> Cube)
3. Ensure infill density is less than 100% and slice
4. On the preview, find the first layer that has an internal bridge and look at the layer below it

### Actual results

I see an extra "solid infill" line that is added inside of the innermost wall. See screenshot for example

[image]

The problem here is that this extra line is printed at solid infill speed over at best a few areas of sparse infill and in the worst case (0% infill) over nothing. The only adhesion it has is to the side wall. That might work if it printed slowly but it tends to default to solid infill speeds which are much faster than overhangs/bridges (generally).

With certain types of filament, like PLA, the issue may not be as noticeable but filaments notorious for lifting/curling, like PETG, this can make some objects almost unprintable without severely lowering speed (have to lower solid infill speed or painstakingly set up layer heigh modifiers for every layer before an internal bridge).

### Expected results

I think the ideal result is to not have this solid infill line at all, like below:

[image]

What would then happen is that the internal bridge layer would simply overlap the innermost wall by a bit, maybe some user set percentage. This would mimic what already is done with sparse infill in terms of infill/wall overlap.

### Project file & Debug log uploads

No project file since it would be more work to download and open it than to recreate it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I know this was introduced by PrusaSlicer and I can understand the argument this may not be a bug since it's operating as intended. But consider the bug is not against the execution of the implementation, but the actual design itself. 

I have yet to be convinced this is a good solution and, at least for me, has caused more problems printing than any other slicer bug in recent memory. This includes when some versions of Orca had incorrectly applied overhang speeds. That was easy for me to work around. This issue here requires me to babysit my printer to prevent nozzle buildup and lines curling up and ruining a print.

That being said, I would say a good initial change that wouldn't go completely against what PrusaSlicer implemented would be to change the speed of this solid infill line to either behave like an overhang or to the bridging speed.

igiannakas, 2025-05-12:
This line is required to anchor the internal bridge. Without it it will fail at the ends as there will be nowhere for the internal bridge to anchor itself. 

I get the argument that we could just ram the bridge to the wall and anchor it there, but then this would cause the wall to over extrude on the next layer, as it will have some material from the bridge overlayed on it, in addition to the wall extrusion. That is undesirable. 

If you're getting failures due to this feature then the below may be the cause:
1. Too low sparse infill density - the above looks like 0%?
2. Too fast internal bridge speed
3. Too fast internal solid infill speed
4. Insufficient cooling / too low minimum layer time thresholds

staal54a, 2025-05-13:
(responding late at night so hopefully this all makes sense but if not, please ask and I can clarify when I'm awake)

Yep, I get why the anchor is placed there which is why I was going back and forth on whether this is a bug. To me, I don't notice any improvement in terms of internal bridging with this being added but I do notice some other minor issues introduced.. For the most part, it usually doesn't result in a failed print and just a little build up on the nozzle. But for large or complex prints, that buildup can start to drop blobs on the print or even cause print failures (again, I have not had a print failure because of this but just noting that could happen).

The screenshot I attached was only for highlighting the line and not an actual use case of mine (though I am now curious how it would print......). When I run into the issue, my settings are somewhat standard (I generally float around 12-15% infill) and it's really only on this anchor, or the sparse infill that is offset by the anchor, which cause me problems with filament not adhering and lifting up. 95% of the time, it just causes that bit to be picked up by the nozzle and the rest of the print goes fine, though the caveat there is that I usually print smaller, less complex items.

Is my printer perfectly tuned? Definitely not. But was I hitting this issue with the same printer (and probably slightly worse tuning) before this anchor line was available? No. I also don't think it's just a problem with me and my printers, but it usually ends up being relatively harmless and so probably wouldn't be high on anyone's priority list to report/request. Plus, I think it would be more prevalent with a filament like PETG (which I would guess I print at higher percentages than most other users) which is more likely to curl up in a given scenario than most other consumer level filament.

I know that this anchor line was intentionally implemented by PrusaSlicer developers to fix some of the issues they were seeing with bridging. So the likelihood of this solution getting reworked is slim here which I knew/accepted going in (I think there's probably a better solution out there that's not too out of the box but I'm not familiar enough with slicers to think of it myself).

My current thought is that if the speed of this anchor line could be changed to either be configurable or to choose a much slower speed than solid infill (maybe bridging speed), that would fix some of the edge cases that are problematic without really introducing any other issues.

igiannakas, 2025-05-13:
Why don’t you post the model file you’re having trouble with including the project file so I can see what you’re seeing?

staal54a, 2025-05-16:
Sorry, was a bit busy with work and other things so I didn't get a chance to respond but it did allow me to dig into this a little further. In terms of project, it's literally just slicing a simple cube against any of the default profiles for any machine. As long as there is a top layer and the infill is <100%, the lines will show up. I generally use 12% cross hatch infill. This issue is also more noticeable/prevalent with PETG/PCTG filament which are more easily pulled up by the nozzle in the case it doesn't stick to a lower layer/wall.

Anyways, based on my testing, it seems that the bridge anchor I was referring to is already printed slower. It's printed at 0.325 * solid_infill_speed. So I was wrong in thinking it was printed at solid infill speed (I don't really print larger parts so a lot of the time my speeds are capped by minimum layer time).

What's interesting and potentially more problematic is that it looks like the line width of that anchor is actually a lot smaller than any other defined line width I have (I use a 0.5 nozzle and generally print with a width around .55-.6 but the bridge anchor line looks like it's closer to 0.4 for some reason). I think that is probably more of an issue in my case.

Anyways, I think I'm going to close this bug for now since I don't necessarily think there is a simple/quick fix to mitigate the issues I'm seeing (probably more of a feature enhancement). I still don't like the implementation by PrusaSlicer per se (or rather I think it could be made more configurable), but it's minor enough that I can work around it somewhat (if I do a large/long PETG print it potentially will leave some blobs but hopefully not a showstopper).

closed 2025-05-16 by staal54a