OrcaSlicer/OrcaSlicer #9568: Print won’t start – Heatbed never reaches target temperature (P1S – OrcaSlicer)

N55PwR, 2025-05-07:
Describe the bug:
When using OrcaSlicer with my Bambu Lab P1S printer, the print process does not start because the heatbed is always reported as 1°C below the target temperature defined in the filament profile. The printer remains stuck at the “preheat bed” stage, even if the actual temperature is extremely close to the setpoint.

This problem does not occur in Bambu Studio. There, the bed can also stay at 1°C below the target value, but the print still starts without issues.

What I’ve already tried:
	•	Lowering the bed temperature in the filament settings → the printer still does not start.
	•	Using G-code variations with M140, M190, and M190 R → no success; the printer remains stuck.
	•	Reinstalling OrcaSlicer and resetting to fresh profiles → no change.
	•	Testing with multiple firmware versions on the P1S → same behavior.

This leads me to believe that OrcaSlicer’s startup logic is too strict when validating the heatbed temperature. Even a 1°C deviation is treated as a reason to block the print from starting, which appears unnecessarily restrictive.

In contrast, Bambu Studio seems to allow a tolerance of at least ±1°C for the bed temperature before initiating the print.

Suggestion:
Please consider adding a tolerance for bed temperature matching in the startup logic (e.g. ±1°C), or allow users to define such a tolerance themselves in the slicer settings or machine G-code.

Printer model:
Bambu Lab P1S

OrcaSlicer version:
[Insert your version, e.g., 1.7.3]

Operating system:
[Insert your OS, e.g., Windows 11]

Firmware version:
[Insert the printer firmware version]

johnny-bit, 2025-05-07:
The behaviour you're describing looks like what happens when one upgrades their P1S firmware... Please check https://github.com/SoftFever/OrcaSlicer/pull/9517 and test it.

Or if you don't have AMS 2 Pro and you don't need the updated FW, just downgrade the firmware to last sane version.

WheelerCC, 2025-05-07:
I saw the same with the 01.08.00.00 firmware on a P1S, downgrading to 01.07.00.00 and the problem goes away. Strangely on one occasion I was able to get the print to start by loading the filament from the AMS before starting the print. I followed https://forum.bambulab.com/t/new-firmware-doent-work-how-do-i-downgrade/47771/2 to downgrade. Orca slicer 2.3.0 build 70931e5

ssozonoff, 2025-05-08:
I have the same issue but its not consistent. P1S and AMS 2 Pro on latest firmware. Orca 2.3.0 build 70931e5
My current workaround is to cancel the job and then reset it from the Bamboo Mobile App using the "history" feature. Doing this and the printing works.

Noisyfox, 2025-05-09:
> I have the same issue but its not consistent. P1S and AMS 2 Pro on latest firmware. Orca 2.3.0 build [70931e5](https://github.com/SoftFever/OrcaSlicer/commit/70931e5321fa66966a5bfb251efca0e82307d427) My current workaround is to cancel the job and then re-send it from the Bamboo Mobile App using the "history" feature. Doing this and the printing works. I cant downgrade because of AMS 2 Pro.

Please give this PR a try https://github.com/SoftFever/OrcaSlicer/pull/9517

Ashot27, 2025-05-17:
I faced with the same problem on my P1S.
To avoid the bug I push "Load" (the "Printer" window in Orca) and after the filament is loaded I start my printing.
It's annoying, but it works.

Noisyfox, 2025-05-24:
> I was all excited to try out the new features in Orca, as I haven't used it in a while, only to find out I have this same issue. The print just won't start. It gets stuck on "preheating bed"

Either downgrade printer firmware or try https://github.com/SoftFever/OrcaSlicer/pull/9517

ThermalX, 2025-05-24:
> > I was all excited to try out the new features in Orca, as I haven't used it in a while, only to find out I have this same issue. The print just won't start. It gets stuck on "preheating bed"
> 
> Either downgrade printer firmware or try [#9517](https://github.com/SoftFever/OrcaSlicer/pull/9517)

I feel like I may be doing something wrong...I don't have the option to toggle legacy plugin. I just have the option "enable network plugin". I thought maybe it was a nightly feature, so I checked the dev build but even there I don't see it. I'd rather not have to downgrade the firmware. Why am I not seeing that toggle option?

Noisyfox, 2025-05-24:
You need to download the PR build.

kazafog, 2025-08-10:
Any other options to fix this for old p1s firmware? 

I am on 01.06.30.01. The PR build did not help regardless of whether the legacy mode was on or off.

kazafog, 2025-08-10:
scratch that, i went back to bambu studio which then indicated a filament retract issue with the ams. Once i fixed this orca then worked. So potentially this is a entirely different issue to do with not reporting a retract issue. I will investigate further

closed 2026-10-30 by github-actions[bot]