OrcaSlicer/OrcaSlicer #9570: Small hole gets filled in when laid on flat

classicrocker883, 2025-05-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Debian Trixie

### Printer

Ender-3 S1 Pro

### How to reproduce

1. Import model
2. Lay/flip so hole touches plate
3. Slice

### Actual results

- Holes are filled in

### Expected results

- Hole to remain a hole, not filled in

### Project file & Debug log uploads

### As you can see, the sliced model fills in the hole.
[image]
---

### Here is how the model is.
[image]
---

### Here is the model sliced 180° (upside down).

[image]
---

<br>

### Zip file.

### [Xiaomi_USB_C_Adapter_Keychan_v2.zip](https://github.com/user-attachments/files/20095435/Xiaomi_USB_C_Adapter_Keychan_v2.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

discip, 2025-05-08:
@classicrocker883
The **`stl`** doesn't help much in understanding the issue you described.
Would you mind uploading the associated project file (**`3mf`**) please.

kisslorand, 2025-05-08:
It's been fixed already.

[image]

closed 2025-08-15 by github-actions[bot]