OrcaSlicer/OrcaSlicer #9573: Core One Profiles not merged

SuperBrunch, 2025-05-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

This bulletin (https://github.com/bwees/orca_coreone/tree/orca_builtin) says the Core One profile has been merged, but the Core One does not show up as an option to select from.

ps I have never used GitHub before sorry if I am doing this wrong

### Printer

Core One

### How to reproduce

click "Select/Remove printers"

### Actual results

Core one is not listed

### Expected results

Core one should be listed

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

johnny-bit, 2025-05-08:
PR https://github.com/SoftFever/OrcaSlicer/pull/9352 was merged last week, while the 2.3.0 release happened before that. You would have to use nightly dev build to get them.

SuperBrunch, 2025-05-16:
> PR [#9352](https://github.com/SoftFever/OrcaSlicer/pull/9352) was merged last week, while the 2.3.0 release happened before that. You would have to use nightly dev build to get them.

Amazing thanks! Got it working :)