OrcaSlicer/OrcaSlicer #9577: support for the Flashforge Creator 3 pro (IDEX dual extruder 3D printer)

Grant-T96, 2025-05-08:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

No problem, just a request to support a widely used printer (Flashforge Creator 3 Pro) that’s currently not on the supported printer list in OrcaSlicer. Users, like myself,  are limited to FlashPrint, which lacks modern slicing features.****

### Which printers will be beneficial to this feature?

Marlin, All, Others

### Describe the solution you'd like

If possible, please can you add the the Flashforge Creator 3 Pro to your current supported printer list. The creator 3 pro is  a professional-grade IDEX 3D printer widely used in design studios and prototyping firms. Its advanced features, such as independent dual extrusion and high-temperature capabilities, make it a valuable addition to the OrcaSlicer supported printers list. Currently, users are limited to FlashPrint, which lacks the advanced slicing features and flexibility that is offered by OrcaSlicer. Incorporating support for the Creator 3 Pro would provide users with enhanced slicing capabilities, improved print quality, and greater control over their printing processes. 

Please find the specs for the printer below:

Printer Name: Flashforge Creator 3 Pro
Manufacturer: Flashforge
Firmware: Proprietary, Marlin-based (closed source)
G-code Flavor: Marlin-compatible (GX format)

**Hardware Specifications**
Build Volume: 300 mm (X) × 250 mm (Y) × 200 mm (Z)

Extruders: Independent Dual Extrusion (IDEX) system

Nozzle Diameter: 0.4 mm (supports 0.6 mm and 0.8 mm)

Maximum Nozzle Temperature: 320 °C

Maximum Bed Temperature: 120 °C

Filament Diameter: 1.75 mm

Compatible Materials: PLA, ABS, PETG, PC, PA (Nylon), PVA, HIPS, ASA, Wood, PA-CF

Connectivity: USB, Wi-Fi, Ethernet, FlashCloud

Touchscreen: 4.3-inch color display

Built-in Camera: Yes
store.p3additive.com

flashforge-usa.com

3D Printing

Geizhals

MatterHackers

[FLASHFORGE](https://www.flashforge.com.hk/creator-3.html?utm_source=chatgpt.com)

FLASHFORGE

12th Man Technology

FLASHFORGE

**Key Features**
IDEX Functionality: Supports mirror, duplicate, and multi-material printing modes

High-Temperature Printing: Capable of printing high-performance materials

Flexible Magnetic Build Plate: Facilitates easy print removal

Auto Bed Leveling: Enhanced leveling accuracy with upgraded extruder sensors

Dual-Fan Configuration: Improves print quality for engineering filaments

Enclosed Chamber: Maintains stable printing environment
FLASHFORGE

center3dprint.com

**Source References**
Flashforge Creator 3 Pro Official Page: 
[FLASHFORGE](https://www.flashforge.com.hk/creator-3-pro.html?utm_source=chatgpt.com)

MatterHackers Product Page: 
[MatterHackers](https://www.matterhackers.com/store/l/flashforge-creator-3-pro-independent-dual-extrusion-3d-printer/sk/MVZ2SZRX?srsltid=AfmBOorAIIG2e0826i5eGvoPdcsWSNepvnsXUXPTXVuaFEtGxBErzYul&utm_source=chatgpt.com)

Flashforge USA Product Page: 
[flashforge-usa.com](https://flashforge-usa.com/products/shop-flashforge-creator-3-pro-independent-dual-extruder-3d-printer?srsltid=AfmBOor-tXM_oveO5DoMyDB4srf9OXvT-zkE6i4rQtbORnN3sdIH3pw4&utm_source=chatgpt.com)

### Additional context

I’m happy to help test this printer with OrcaSlicer if needed and can provide sample G-code or configuration details where possible.

Thank you for your consideration and the continued development of OrcaSlicer.

davaze, 2025-08-24:
Same here, owning a FF Creator 3 v2. Tried with FF support to get a working profile but it seems they are not interested in providing anything.

ingoknito0815, 2025-10-01:
I work for a company with a creator 4 and got the same problem. Hardware is good software is trash, specifically the slicer (Flashprint)
I think a working Creator 3 Profile would also help als Ceator 4 Users.
I tried a little bit but i could only get the right extruder to work a bit.
The main problem seems to be the conversion from gcode to gx.
There is a python program by the github user urself25 called Orca_GCode_to_Gx wich helped a little bit but sadly not enough so far.

I also contacted FF multiple times and it seems like they do not really care at all.

jdamon11, 2026-03-24:
@FlashforgeOfficial How about some support for the Creator/Creator 3 Pro?  Especially seeing that Flashprint doesn't have idle/standyby temperature setting making dual filament printing useless unless do mirror or duplicate mode.

Felix14-v2, 2026-03-25:
Wow, that's really cool! But almost 400 thousands of polygons sounds crazy. Could you send it in STEP/IGES? I'd tweak the model to optimize its final size (like replace some hidden fillets with chamfers or moving text to the texture layer).

> will add the texture/colors by .png later

Can we use .svg? It will make further changes much easier, especially when we get to a testing stage. IDEX modes and the corresponding print area should be marked there, and I'm sure that some tweaking will be needed.

ptwittmann-hash, 2026-03-26:
> > Is there a orca slicer profile at the moment?
> 
> Only with single material setup. I'm working on tool changing logic right now, but it still needs some polishing (and workarounds for the new OrcaSlicer bugs introduced in 2.3.2) before moving on to the testing phase. I think I'll finish this over the weekend. IDEX modes should work too, but I never tested them.

How can we help you?
Is the single profile tested?

Felix14-v2, 2026-03-30:
After I took a look at Creator 3, it turned out that it differs from Creator 2 a lot more than I expected. So I focused on solving remaining problems with Creator 2 and trying to make IDEX Copy/Mirror modes work (it's 95% done, just some required metadata is missing). But single extrusion (1 or 2 colored prints) mode should work.

### Features
- Single nozzle printing
- Dual-nozzle printing (alternating)
- Per-extruder placement checks
- Basic toolhead collision checks
- Basic purging trick to improve towerless tool swapping (Creator Pro 2 doesn't have nozzle cleaners, so its reliability is questionable and requires testing)
- Backwards compatibility with FlashPrint (line type macro makes FlashPrint correctly recognize and color lines)

### Installation (Windows)
1. Download [Creator Pro 2 – OrcaSlicer profile v0.01.zip](https://github.com/user-attachments/files/26336334/Creator.Pro.2.OrcaSlicer.profile.v0.01.zip)
2. Unzip it
3. Drag the `user` folder and drop it to the `%AppData%/OrcaSlicer` folder
4. Restart OrcaSlicer
5. [Optional] Load the bed texture in the printer profile settings (Printer settings → Basic information → Printable space → Printable area → `Set ...` → Texture → `Load ...`, go to `%AppData%/OrcaSlicer/user/default/machine/base` and select `Flashforge Creator Pro 2.svg`)

### Testing & feedback
Please perform the following calibrations:
[Filament-specific]
- Flow ratio test
- Maximum flowrate test (use this 3MF to generate it: [Flowrate calibration (10-20 mm3 per second).zip](https://github.com/user-attachments/files/26336244/Flowrate.calibration.10-20.mm3.per.second.zip))
- Retraction test
- Overhang test (you can use any preferred model, just don't forget to leave a link)

[Hardware & firmware]
- Cornering test
- Pressure advance test (does the printer support it?)
- VFA test
- Relative extrusion (try to print something with `Use relative E distances` enabled in the printer profile settings)

Please, share your feedback [in Discord](https://discord.com/channels/1137181739773603922/1139519763077406830) (don't forget to ping `@felix14-v2`)

## WARNING: this profile is unofficial, extremely experimental and requires a lot of testing (I can't test it myself because I don't have any FlashForge printers). Please proceed with caution!

Felix14-v2, 2026-04-01:
Oh, and also big thanks to @ssalar44 for the bed model for Creator 3! I spent some time optimizing it as much as possible, and managed to reduce its size to <500KB without big loss in quality. I can't embed it to the profile, so you have to manually load it in the printer profile settings similar to the texture loading method described above).

[Flashforge Creator 3 Pro.zip](https://github.com/user-attachments/files/26423739/Flashforge.Creator.3.Pro.zip)

I still haven't looked into the texture unfortunately, but I still don't know actual left/right nozzle printing area dimensions. It would be really helpful if someone measured them and made a drawing [like this](https://wiki.evilazrael.de/en/flashforge-creator-pro-2#advanced-configuration-extended-print-bed), but for Creator 3 Pro.

umky, 2026-04-07:
> made a drawing [like this](https://wiki.evilazrael.de/en/flashforge-creator-pro-2#advanced-configuration-extended-print-bed), but for Creator 3 Pro

Here you go.

[image]

ptwittmann-hash, 2026-04-08:
Okay, everyone, I'm currently testing Felix's Orca profiles, and so far everything is going well. Maybe a few more of you could give it a try. So far, I've tested single-extruder mode and Idex mode (two filaments, one of which is for the support structure). What I still need to test is mirror mode.
I think we need an experienced user to check if there are any errors or if everything is running smoothly.

Felix14-v2, 2026-04-16:
> Within the machine settings extruder_printable_high (205 mm) as well as extruder_printable_area (for extruder 1: "-160x-125,145x-125,145x125,160x125") can't be edited to the proper dimensions.

Sorry for the long reply. You can just remove these settings from the profile by editing it in notepad or other text editor (if you don't have an IDE). If you are on Windows, the profile should be in the `OrcaSlicer/user/default/machine/base` folder. If you have another folder near the `default` one, you should move the `default`'s folder content into it.

HellEvro, 2026-09-25:
For anyone still blocked on **G-code → `.gx`** (thumbnails / printer UI metadata) when using Orca with Creator 3 Pro / dual-extruder Flashforge:

I published an Orca post-processing converter with explicit printer profiles (including `creator3pro` dual):

- Repo: https://github.com/HellEvro/FF_Gcode_to_GX
- Binaries (Win / macOS arm64 / Linux): https://github.com/HellEvro/FF_Gcode_to_GX/releases/tag/v0.1.0

Setup is the usual Orca path: filename `*.gx` + post-processing script pointing at the binary (`--printer creator3pro` optional). Dual toolchanges still belong in the Orca machine profile — the converter only wraps the Flashforge header + BMP.

Hardware feedback welcome in the repo Issues.

HellEvro, 2026-09-25:
**Update — Creator 3 Pro Orca machine G-code** (optional start / pause / end-placeholder):

https://github.com/HellEvro/FF_Gcode_to_GX/tree/main/profiles/creator3pro

Community help wanted:

- Share / improve settings: https://github.com/HellEvro/FF_Gcode_to_GX/discussions/3
- Request: best full Orca profile: https://github.com/HellEvro/FF_Gcode_to_GX/discussions/4
- Request: filament presets (PLA/PETG/ABS/TPU/…): https://github.com/HellEvro/FF_Gcode_to_GX/discussions/5