OrcaSlicer/OrcaSlicer #9580: Crash on slicing complicated model

RX14, 2025-05-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Printer

Bambu Lab P1S

### How to reproduce

1. Download the attached project
2. Slice it

### Actual results

Crash at "Detect overhangs for auto-lift"

### Expected results

A sliced gcode

### Project file & Debug log uploads

[Energy-Buran.zip](https://github.com/user-attachments/files/20103981/Energy-Buran.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Here's a backtrace:
```
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x000072c5a9ea7813 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:89
#2  0x000072c5a9e4ddc0 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x000072c5a9e3557a in __GI_abort () at abort.c:73
#4  0x000072c5b2c9a421 in std::__glibcxx_assert_fail (file=file@entry=0x573b38c03b18 "/usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/include/c++/optional", 
    line=line@entry=475, 
    function=function@entry=0x573b38c48400 "constexpr _Tp& std::_Optional_base_impl<_Tp, _Dp>::_M_get() [with _Tp = unsigned int; _Dp = std::_Optional_base<unsigned int, true, true>]", condition=condition@entry=0x573b38d766b7 "this->_M_is_engaged()")
    at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
#5  0x0000573b3614aaa0 in std::_Optional_base_impl<unsigned int, std::_Optional_base<unsigned int, true, true> >::_M_get (this=<optimized out>)
    at /usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/include/c++/optional:473
#6  std::_Optional_base_impl<unsigned int, std::_Optional_base<unsigned int, true, true> >::_M_get (this=<synthetic pointer>)
    at /usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/include/c++/optional:473
#7  std::optional<unsigned int>::operator*() & (this=<synthetic pointer>) at /usr/lib/gcc/x86_64-pc-linux-gnu/14.2.1/include/c++/optional:972
#8  Slic3r::solve_extruder_order (wipe_volumes=std::vector of length 1, capacity 1 = {...}, 
    all_extruders=std::vector of length 1, capacity 1 = {...}, start_extruder_id=std::optional [no contained value])
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/libslic3r/GCode/ToolOrdering.cpp:41
#9  Slic3r::get_extruders_order (wipe_volumes=std::vector of length 1, capacity 1 = {...}, 
    all_extruders=std::vector of length 1, capacity 1 = {...}, start_extruder_id=std::optional [no contained value])
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/libslic3r/GCode/ToolOrdering.cpp:98
#10 Slic3r::ToolOrdering::reorder_extruders_for_minimum_flush_volume (this=0x72c3737f69a0)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/libslic3r/GCode/ToolOrdering.cpp:914
#11 0x0000573b38326c60 in Slic3r::ToolOrdering::ToolOrdering(Slic3r::Print const&, unsigned int, bool) [clone .constprop.0] (this=0x72c3737f69a0, 
    print=..., prime_multi_material=false, first_extruder=4294967295)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/libslic3r/GCode/ToolOrdering.cpp:310
#12 0x0000573b35f46091 in Slic3r::Print::process (this=0x573b80de5c70, time_cost_with_cache=0x0, use_cache=<optimized out>)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/libslic3r/Print.cpp:2068
#13 0x0000573b369e815a in Slic3r::BackgroundSlicingProcess::process_fff (this=0x573b7a8f9930)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/slic3r/GUI/BackgroundSlicingProcess.cpp:229
#14 0x0000573b369ea5af in Slic3r::BackgroundSlicingProcess::call_process (this=0x573b7a8f9930, ex=...)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/slic3r/GUI/BackgroundSlicingProcess.cpp:439
#15 Slic3r::BackgroundSlicingProcess::thread_proc (this=<optimized out>)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/slic3r/GUI/BackgroundSlicingProcess.cpp:336
#16 0x0000573b369eae54 in Slic3r::BackgroundSlicingProcess::thread_proc_safe (this=<optimized out>)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/slic3r/GUI/BackgroundSlicingProcess.cpp:482
#17 operator() (__closure=<optimized out>) at /usr/src/debug/orca-slicer-git/orca-slicer-git/src/slic3r/GUI/BackgroundSlicingProcess.cpp:522
#18 boost::detail::thread_data<Slic3r::BackgroundSlicingProcess::start()::<lambda()> >::run(void) (this=<optimized out>)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/deps/build/destdir/usr/local/include/boost/thread/detail/thread.hpp:120
#19 0x0000573b36ddc24d in boost::(anonymous namespace)::thread_proxy (param=<optimized out>)
    at /usr/src/debug/orca-slicer-git/orca-slicer-git/deps/build/dep_Boost-prefix/src/dep_Boost/libs/thread/src/pthread/thread.cpp:177
#20 0x000072c5a9ea57eb in start_thread (arg=<optimized out>) at pthread_create.c:448
#21 0x000072c5a9f2918c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

The crash appears to be if `solve_extruder_order` is called with an empty `start_extruder_id`. From reading the `std::optional` documentation, the expression `*start_extruder_id = all_extruders.front();` where `start_extruder_id` is an `std::optional` which doesn't contain a value may be undefined behaviour, in this case causing an assert.

RX14, 2025-05-08:
It doesn't appear that putting an item in an `std::optional` that's already empty is easy: https://devblogs.microsoft.com/oldnewthing/20241113-00/?p=110516

Noisyfox, 2025-05-09:
I think the issue is the `*`. I'll see if `start_extruder_id = all_extruders.front();` works.

PR #9584 (2025-05-09): Fix crash when `start_extruder_id` is empty
Fix #9580
closed 2025-05-09 by Noisyfox