OrcaSlicer/OrcaSlicer #9583: Multi material wall order

nil80c, 2025-05-09:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When printing with multiple materials, wall order is ignored when laying down the secondary filament.
Example,
Create a cube with filleted edges.
Paint the fillets with a different filament.
Chose inner / outer wall order.
Printer will print the primary filament correctly, but always print the secondary filament outer wall first causing overhang issues.

[image]
[image]
[image]
[image]

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

It should print the walls in the order picked.

nil80c, 2025-05-09:
Pictures of primary wall.

[image]
[image]

Noisyfox, 2025-05-09:
The painted part has only single wall, so the wall order does not make any difference.

nil80c, 2025-05-09:
> The painted part has only single wall, so the wall order does not make any difference.

Respectfully I disagree. 
In my first post, if you look at the first picture the primary colour is already laid with 3 walls and infill, which is blue. All walls printed in correct order, as shown in my second post. 
It also shows the first wall  of the secondary colour, orange, being laid with a gap to the primary colour as its printing the outer wall first which has an overhang. 
The second and third pictures show the inner wall, then the middle wall being printed. 

This is causing layer adhesion issues on my overhang. 

On inner fillets it's fine as no overhang.

Noisyfox, 2025-05-09:
> The second and third pictures show the inner wall, then the middle wall being printed.

There is no such inner wall for the second color, just switch to line type mode and you'll see.

nil80c, 2025-05-09:
> > The second and third pictures show the inner wall, then the middle wall being printed.
> 
> There is no such inner wall for the second color, just switch to line type mode and you'll see.

Ok, thanks for the info, will check when I get home. 
Apologies if I've wasted your time. And mine tbh, I was up till 2am messing with settings trying to find a solution. 
It's strange that in filament mode it shows it drawing 3 lines. 
Is there a reason for that or just a quirk of the software?

Noisyfox, 2025-05-09:
> Is there a reason for that or just a quirk of the software?

That's the limitation of the app. It requires every wall to be a closed loop.

nil80c, 2025-05-09:
> > Is there a reason for that or just a quirk of the software?
> 
> That's the limitation of the app. It requires every wall to be a closed loop.

Seems I'm on a steep learning curve. Thanks for taking the time to educate. 👍

igiannakas, 2025-05-15:
You can also try printing with Arachne. In cases like this it is more effective in creating appropriate wall ordering as either the gap fill is converted to a thin wall (and is counted as such, hence printed first) or is omitted completely and the walls made thicker.

nil80c, 2025-05-15:
> You can also try printing with Arachne. In cases like this it is more effective in creating appropriate wall ordering as either the gap fill is converted to a thin wall (and is counted as such, hence printed first) or is omitted completely and the walls made thicker.

That's a great tip thank you. 
I had wondered at the use of Archane. 
I'll try that on my next print. 👍

I see what Noiseytfox was saying now I've looked at it more. 
The second color is not printing an inner and outer wall, just a wall in general. 
But where in some places where it was looping to create the correct thickness, it was doing the outer loop first which I thought was strange.

igiannakas, 2025-05-15:
In my view unless you have a specific reason to use classic perimeter generation I’d steer you towards Arachne. It’s a more “modern” approach which saves a lot of time during printing plus offers great quality. 

There are some limitations and exceptions to this but unless you know you need them you’re better served with Arachne. 

For example classic always guarantees a perimeter line width vs Arachne where it’s variable within certain limits. In most cases that is desirable but there may be some engineering models (usually) that need precise control of extrusion widths. But if you don’t know you need it, then you most likely don’t…

igiannakas, 2025-05-15:
What you are seeing is gap fill printed which always comes last. With Arachne there is no gap fill (mostly). - it’s converted to walls.

nil80c, 2025-05-15:
Great explanation. 
It may not be suitable for my parts in that case, but I'll be running some test prints over the weekend to check. 
Thanks again.

ianalexis, 2026-10-14:
In top of what Noisy and Igiannakas wrote, you can set Tool/color order now, still different that what you asked for but related https://www.orcaslicer.com/wiki/print_settings/multimaterial/multimaterial_settings_advanced.html#toolchange-ordering

closed 2026-10-14 by ianalexis