OrcaSlicer/OrcaSlicer #9586: Opening 3mf for Bambu H2D crashes after profile switched to Neptune 4

digitaldragonfire, 2025-05-09:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10/11

### Additional system information

CPU: Intel Core i7-7700K @ 4.2GHz
Memory: 32GB
Graphics Card: Radeon RX 580 Series (8GB)

### Printer

Elegoo Neptune 4

### How to reproduce

1. Loading the 3mf for this model: https://makerworld.com/en/models/1353938-carpenter-bee-trap-fits-mason-jar-or-soda-bottle?from=search#profileId-1397156
2.  Switch the printer from H2D to Neptune 4.

### Actual results

Orca crashes after ~15 to 30 seconds. No exception message, just a message that Orca has stopped working. Also noticed that the model will flash as the mouse is moved.

### Expected results

Orca should not crash.

[crash_Fri_May_09_13_01_14_0.log](https://github.com/user-attachments/files/20126062/crash_Fri_May_09_13_01_14_0.log)

### Project file & Debug log uploads

[BeeTrapProject.zip](https://github.com/user-attachments/files/20126057/BeeTrapProject.zip)

[crash_Fri_May_09_13_01_14_0.log](https://github.com/user-attachments/files/20126086/crash_Fri_May_09_13_01_14_0.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

Noisyfox, 2025-05-11:
Could you send me your entire ~/appdata/roaming/Orcaslicer folder? Thanks

Noisyfox, 2025-05-11:
Btw is your neptune4 profile happen to have only a single filament added? Ie:

[image]

digitaldragonfire, 2025-05-12:
Yes, only one filament slot is available. When I select my P1P profile it shows multiple slots as I would expect.

 

Ken Russell

✉         ***@***.***> ***@***.***