OrcaSlicer/OrcaSlicer #9591: Dynamic Pressure Advance Based on Travel Distance

Caramel1966, 2025-05-10:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

🧵 Feature Request: Dynamic Pressure Advance Based on Travel Distance
Summary
When printing with a long melt zone (e.g. 50 mm melt chamber), extrusion latency becomes significant — especially after long non-extrusion moves (travel moves). A static Pressure Advance (PA) value is not sufficient to compensate across all move lengths.

This request proposes a dynamic Pressure Advance, where the PA value is scaled proportionally to the length of the last travel move before extrusion.

Why It Matters
For hotends with long melt zones (e.g. SuperVolcano, custom melt chambers), the pressure recovery time after a travel move is not constant.
In this context:

Short travel (<10 mm) needs little to no compensation.

Long travel (~150 mm) leads to severe under-extrusion at the start of the line if PA isn't significantly increased.

A static PA leads to one of two compromises:

Either short travels over-extrude (if PA is tuned for long travel),

Or long travels under-extrude (if PA is tuned for short travel).

Proposed Implementation
Introduce an optional dynamic PA system, controlled via advanced settings.
Concept:

Track the last travel distance before any extrusion command.

Apply a Pressure Advance value proportional to that distance:

yaml
Copier
Modifier
if travel_distance >= max_travel:
    PA = PA_max
else:
    PA = PA_max * (travel_distance / max_travel)
Example:

PA_max = 0.06, max_travel = 150 mm

A 75 mm move would yield PA = 0.03

A 10 mm move yields PA = 0.004

This can be implemented as an optional G-code post-processing step, or integrated into OrcaSlicer’s move planner.

Benefit
Solves early under-extrusion issues in long-travel, high-inertia setups.

Improves print quality for large-format printers, dual extrusion systems, and high-flow heads.

Offers a precision tuning mechanism without relying on macros or external scripts.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Example:

PA_max = 0.06, max_travel = 150 mm

A 75 mm move would yield PA = 0.03

A 10 mm move yields PA = 0.004

### Describe alternatives you've considered

Example:

PA_max = 0.06, max_travel = 150 mm

A 75 mm move would yield PA = 0.03

A 10 mm move yields PA = 0.004

Caramel1966, 2025-05-10:
🔧 Example Configuration Showing Workaround Limitations
To achieve a balanced result and avoid under-extrusion after long travels, I currently rely on a fixed Pressure Advance (0.05) combined with a restart extra (2.5 mm) and a minimum travel threshold for retraction (80 mm).

However, this is a compromise:

Short travels → excessive restart

Long travels → not enough pressure recovery

See attached screenshots from my OrcaSlicer filament profile:

🖼️ Screenshot 1: [Filament PA settings]
🖼️ Screenshot 2: [Retraction & restart config with 80mm threshold]

These illustrate the need for a dynamic Pressure Advance scaling based on previous travel distance.
A post-processed G-code sample is also attached where SET_PRESSURE_ADVANCE is adjusted automatically.

[image]

[image]

xantrk, 2025-05-12:
I have a feeling this is more of a klipper/pressure advance issue since early days were focused on 0.4 nozzles this wasn't a problem. But would love to have a workaround in slicer for this and totally agree with problem and proposed solution as a 0.6 nozzle user