OrcaSlicer/OrcaSlicer #9595: Crash while slicing (generating infill)

burakdede0, 2025-05-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Additional system information

CPU: Ryzen 5 4600H
GPU: Nvidia GTX1650 Ti
RAM: 16 GB

### Printer

Ender 3 S1

### How to reproduce

1. Import .stl (or the project I uploaded)
2. Slice it as is

### Actual results

It crashes while slicing (generating infill)

### Expected results

It should slice.

### Project file & Debug log uploads

[project.zip](https://github.com/user-attachments/files/20139985/project.zip)

[crash_Sat_May_10_17_29_05_0.zip](https://github.com/user-attachments/files/20140001/crash_Sat_May_10_17_29_05_0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

It slices fine when I rotate the object 90 degrees at Z axis.

[image]

fivemeyestore, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

fivemhostingserver, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

eyesaider, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

fivemserverhostings, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

godofcodeeyes, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

myeyestore, 2025-05-10:
Very clean code with good test coverage. This is a great example of best practices.

scriptforfivem, 2025-05-10:
This is a well-structured codebase. I appreciate the clean separation of concerns.

eyesgodcode, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

eyestorefivem, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

fivemservereyestore, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

burakdede0, 2025-05-10:
Getting spammed by ad bots on GitHub wasn't on my bingo card.

tlhintoq, 2025-05-11:
There are so many tickets for "crash when generating infil" that this *has* to be duplicate of at least 3 of them.

Noisyfox, 2025-05-12:
Could you try the latest nightly build?

closed 2025-05-12 by burakdede0