OrcaSlicer/OrcaSlicer #9602: PR  "Color & shape improvements for Axes #9495" breaks brim painted drawing

vovodroid, 2025-05-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Win10x64

### Printer

Any

### How to reproduce

PR  "Color & shape improvements for Axes #9495" breaks brim painting.
Add object, try to paint brim.
Reverting this PR fix the problem.

### Actual results

Cursor at the yellow arrow:
[image]

Brim shown in various locations, depending on zoom and view angle:
[image]

### Expected results

Brim is always shown in correct place:
[image]

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-05-12:
Hmmmm I could reproduce this only when looking upwards from bottom of the plate.

Noisyfox, 2025-05-12:
@yw4z could you take a look at this issue?

kisslorand, 2025-05-12:
For me as soon as I close the "Brim Ears" gizmo they disappear regardless if they were manually painted or auto-generated. If I open the gizmo than they reappear.

Noisyfox, 2025-05-12:
> For me as soon as I close the "Brim Ears" gizmo they disappear regardless if they were manually painted or auto-generated. If I open the gizmo than they reappear.

This is expected.

vovodroid, 2025-05-12:
@Noisyfox 
> Hmmmm I could reproduce this only when looking upwards from bottom of the plate.

Issue depends on bed size and object placement. I added sample project.

Noisyfox, 2025-05-12:
Ok I've found the issue. It's an issue in the mouse brim gizmo. Will raise a PR soon.

PR #9605 (2025-05-12): Fix brim ear render when camera projection changed
Fix #9602
closed 2025-05-12 by Noisyfox