OrcaSlicer/OrcaSlicer #9617: Perimeters over sparse infill are only supported on the outer wall

Nukle0n, 2025-05-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 Dev (And older)

### Operating System (OS)

Windows

### OS Version

11

### Printer

Sovol SV06 Plus Klipper

### How to reproduce

Slice a model with perimeters that start above sparse infill

### Actual results

The perimeters are mostly printed at regular speed over the sparse infill, resulting in pinholes and prints that are more permeable.

The preceding layer
[image]

The layer above this

[image]

Note how only the outermost wall (orange) is supported, the rest are free to sag into the infill.

### Expected results

Ideally there would be some bridging underneath the entire perimeter, or the "top" layer would be expanded so the entire perimeter sits on it.

### Project file & Debug log uploads

[AM4 Trayx.zip](https://github.com/user-attachments/files/20191465/AM4.Trayx.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

igiannakas, 2025-05-14:
[image]

This will disable extra alternate walls, which you don't really need in this model.

Nukle0n, 2025-05-14:
> [image]
> 
> This will disable extra alternate walls, which you don't really need in this model.

Yes, that's true. I guess the bug is with alternate extra wall and not with the slicing in general. That's unfortunate since I really like that feature and it's a shame that this makes it unusable for many models with walls in the middle of top surfaces.

Nukle0n, 2025-05-14:
> Not exactly. The vertical shell thickness needs to be in one of its reduced mode as the alternate extra wall will create walls that are overhanging internally, which if enabled in full would cause solid infill to be extruded in the area of the missing alternate wall.
> 
> In any case the alternate wall feature is helpful really to better anchor the sparse infill to the object and that on very specific occasions. So it’s best to generally have it disabled and enable by exception only when the model needs it.

I like it for the extra strength this provides. It would be fine if it was just the 1 perimeter that would maybe overhang the infill, but right now it's all of the internal perimeters that don't get support. It's true not all models need it but I like to do 2-3-2-3 instead of 3-3-3-3 for that reason, and it's really a shame that a slicing bug prevents this from working in a reasonable manner.

closed 2025-08-21 by github-actions[bot]