OrcaSlicer/OrcaSlicer #9618: Exhaust fan turns on at zero percent

bgswiss1, 2025-05-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

23H2

### Printer

qidi Q1 pro

### How to reproduce

1. set filament Exhaust fan and auxiliary fan to zero
2. Start pressure advance test
3. Watch exhaust fan turn on after purge line
4. Watch print fail after 12 layers

### Actual results

Print fails due to too much wind.

### Expected results

I expect the exhaust fan to not turn on when I put it to 0% this is what the included slicer with my machine does but that's the worst software I've ever used as it doesn't have good calibrations(only comes with 3)

### Project file & Debug log uploads

[failed Pressure Advance Test orca fan fail.zip](https://github.com/user-attachments/files/20194633/failed.Pressure.Advance.Test.orca.fan.fail.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

Noisyfox, 2025-05-15:
I didn't see any gcode for starting auxiliary or exhaust fan slicing the given project.

bgswiss1, 2025-05-16:
[bbswheel0exfan.zip](https://github.com/user-attachments/files/20237381/bbswheel0exfan.zip)

bgswiss1, 2025-05-16:
I'm not sure if this helps but when I open the orca Project file in qidi studio the option for ex fan percentage is not show just to toggle on and off

bgswiss1, 2025-05-16:
[image]

Noisyfox, 2025-05-16:
The gcode from qidi, not the project file.

bgswiss1, 2025-05-16:
[workingexfan.zip](https://github.com/user-attachments/files/20237473/workingexfan.zip)

bgswiss1, 2025-05-16:
Also i have been using orca with these settings working prior. After trying the pressure advance test the exhaust fan keeps coming on now no matter what before then if I set it to zero it wouldn't turn on.

Noisyfox, 2025-05-16:
Could you try adding a `M106 P3 S0` at the end of the print start gcode and see if that solves the problem?

bgswiss1, 2025-05-16:
this worked but The exhaust fan does still turn on in the beginning, it turns off after the purge line with this

Noisyfox, 2025-05-16:
Then maybe try add this line to the beginning?

Noisyfox, 2025-05-16:
[image]

Left qidi, right Orca

The only difference is the missing `M106 P3 S0` from Orca. I can't really understand why they shouldn't work identical once we added the missing `M106 P3 S0` after the purge line completed.

Noisyfox, 2025-05-16:
Or maybe is the missing `M141 0` that triggers the exhaust fan? Could you try adding this at the same place as qidi?

bgswiss1, 2025-05-19:
I think that is the problem. For the end gcode mine has the M141 S but not for the start.
thank you

bgswiss1, 2025-05-19:
I consider the issue some but I'll leave it open in case you guys need to do something with it thank you

bgswiss1, 2025-05-19:
sorry working backwards but the  M141 S0 did turn on the fan for a few seconds when put in the qidi start gcode

closed 2025-08-26 by github-actions[bot]