OrcaSlicer/OrcaSlicer #9621: More cooling option - top and external perimeter fan speed

locki-cz, 2025-05-13:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

This make nicer prints, looks nice outside and strong bonds due to more cooling options for visible external perimeters and possible to disable on top layers.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Possible disable cooling for top layer, enable moore cooling for external perimeter only looks also better, possible merge it from SuperSlicer:: [https://github.com/supermerill/SuperSlicer/pull/3754/commits/09ba8b8449dc6c3f2ffb9482c9e6239407abf1f9](https://github.com/supermerill/SuperSlicer/pull/3754/commits/09ba8b8449dc6c3f2ffb9482c9e6239407abf1f9)

locki-cz, 2025-05-14:
It is possible to implement it? @vovodroid ;)

vovodroid, 2025-05-15:
External perimeters are here:

[image]

vovodroid, 2025-05-15:
Are you sure top cooling is necessary?

locki-cz, 2025-05-15:
> Are you sure top cooling is necessary?

yes :)

locki-cz, 2026-10-20:
@Felix14-v2 is there any progres on that in 2.5.0 version?

Felix14-v2, 2026-10-20:
I'm just a regular user, not developer. But I think you already can do that using filament cooling settings:

[image]

[image]