OrcaSlicer/OrcaSlicer #9623: Random floating bottom layers in gcode that aren't model geometry

tlhintoq, 2025-05-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 dev build 0cb458d

### Operating System (OS)

Windows

### OS Version

Win10x64

### Additional system information

[image]

### Printer

Elegoo giga

### How to reproduce

Sliced a fairly tall object

### Actual results

In viewing the gcode I noticed a bunch of random bottom layer segments that seem to come diagonally from corners - where there is no such feature in the model.  Yes the models have been checked for problems (using 3d builder which is where that screenshot comes from)

[image]

### Expected results

Don't just add in random floating bits to cause problems - or to get knocked loose inside a basically hollow print to become a noisy rattle with no way to pull out.

### Project file & Debug log uploads

[HeavyGun.giga.orca.zip](https://github.com/user-attachments/files/20197388/HeavyGun.giga.orca.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

These almost look like they're meant to be corner bracing because they seem to always come up as diagonals at perpidicular walls.  But as you can seen they aren't features in the model.

tlhintoq, 2025-05-14:
I just updated to nightly build 1a5b238
Still does it.

[image]

tlhintoq, 2025-05-14:
Digging deeper...
Looks like this random useless stuff has been  happening for years.  There's old bugs with other names.  Reddit threads with people trying to kill it.  So clearly its been a nuscance for years.
https://www.reddit.com/r/OrcaSlicer/comments/1chwfnj/anyway_to_prevent_these_random_bits_of_internal/
https://github.com/SoftFever/OrcaSlicer/issues/2842

Maybe now is the time to offer a check box to just turn off internal bridging?  Or some kind of slider for adjusting the size of internal bridge features that starts at 0 

But - what's the point of all this random bridging but to eat time, material and potentially cause problems for a nozzle to get hung up on?

https://github.com/user-attachments/assets/5970d10b-3402-485b-8996-9c8cd7963a7d

igiannakas, 2025-05-14:
This happens because you need a bridge layer before solid layers are laid down to enable you to print your top surfaces. 

The bridge doesn't have much to anchor to because your sparse infill density is too low and there is no sparse infill to anchor against. Hence it's failing. 

[image]

If you enable a longer sparse infill anchor then the bridge infill will have somewhere to anchor to. No need to increase the infill density but rather tweak the anchor size to enable the bridge to have a place to sit on.

[image]

Removing this is not and should not be an option as it is needed for the solid layers to lay on. The settings changes should focus on getting some sparse infill underneath the bridge to let it reliably anchor itself.

shpakovkv, 2025-05-16:
Faced with a similar problem - internal bridges that completely overlap the holes (even huge ones) in the places of stepwise diameter change. Moreover, when I looked at the height of the layer, it turned out that this bridge is part of the model, not the support. But the model has a hole there, and no thin layers. 

In my case, changing the "Bridge counterbore holes" parameter to "None" + using manual supports solved my problem.

tlhintoq, 2025-05-17:
> This happens because you need a bridge layer before solid layers are laid down to enable you to print your top surfaces.
If you check the video from going through the layers you'll see there are no top surfaces above these bridges.  They serve no purpose and are supporting nothing.

> Removing this is not and should not be an option as it is needed for the solid layers to lay on. 
If this were happening below a top layer I'd agree with you.  But its not.  This is maybe 20% of the way up from the bed with no top surfaces for at least 250mm in z above this point.

igiannakas, 2025-05-17:
It’s probably because of solid infill drawn due to he ensure vertical shell thickness function - try setting this to one of the two reduced modes.

tlhintoq, 2025-05-18:
> It’s probably because of solid infill drawn due to he ensure vertical shell thickness function - try setting this to one of the two reduced modes.

I'm trying to wrap my head around how you arrived at that connection.  Not arguing it, just trying to see how you got there.
The `bridging` in an empty voidcould be from `infil` because of `ensure shell thickness`....  Huh?  Ok, I'll see what I can do with that.

> try setting this to one of the two reduced modes.
It was already at `Critical only` for those earlier posts.  That seems pretty minimal.
---
I just set it to `none` - that's about as reduced as there is, right?
ummm... maybe some differences.  Not sure I'd call it 'less', or that it stopped doing it where it wasn't needed or did more where it could be needed below a bottom surface... but more like it just did it differently.
---
Let's see what 'all' makes it do.  If you're right there should be a truly visible difference from `None` to `All`.
Doesn't look any different than 'none'.
---
Ok... what' about `moderate`?
Ehhh... Again maybe a little different but nothing that seems significant to like I can point to any one instance of feature and say "oh that's better/worse"
---

I guess I'll leave the vertical shell at `none` until I see a reason to have it on.  Less math and less playing with things is better until I can point to a reason to have it.

I appreciate the input and the 'out of the box' thinking!

igiannakas, 2025-05-18:
None on vertical shell thickness will lead to top surfaces failing if there is even the slightest hint of a slant. So not recommended.

In your posted model above can you please point to me which one of the plates you’re referring to and what layer? So I can troubleshoot it and guide you better

closed 2025-08-25 by github-actions[bot]