OrcaSlicer/OrcaSlicer #9624: Can not generate Tree Support on my customized voron2.4 profile, neither auto nor manual .

Jazzxianren, 2025-05-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 and also 2.3.1 - dev build 41ff632

### Operating System (OS)

Windows

### OS Version

windows 11

### Printer

Voron 2.4

### How to reproduce

1. normal support generation is ok, both auto and manual

2. can not generate tree support, neither auto nor manual

### Actual results

[image]

### Expected results

no tree support can be generated

### Project file & Debug log uploads

[tree support issue report.zip](https://github.com/user-attachments/files/20202757/tree.support.issue.report.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-05-15:
It's the issue of your excluded bed area settings. Currently it will cause unexpected behavior to have concave shaped exclusion area.
Just remove the excluded bed area settings and it should work.

Noisyfox, 2025-05-15:
Please give https://github.com/SoftFever/OrcaSlicer/pull/9633 a try

closed 2025-06-01 by Noisyfox