OrcaSlicer/OrcaSlicer #9627: Can the orca support print from the position close to the nozzle in new layer?

hackqiang2, 2025-05-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I'm not sure if I missed the relevant configuration, so I'm here to seek help. Thank a lot.
For example, when I print a long snake, each layer is shot from the snake head to the snake tail, that is, after each layer is finished, toolhead has to move to the snake head to start print the next layer. This empty driving process wastes time.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I think it's better to start printing directly from the position close to the nozzle in the new layer.

MxBrnr, 2025-05-14:
This is intentional because the filament needs time to cool and solidify before more can be put on top of it. PrusaSlicer, Bambu Studio, OrcaSlicer, Cura, and any of their derivatives all act this way.

hackqiang2, 2025-05-14:
got it.

closed 2025-05-14 by hackqiang2