OrcaSlicer/OrcaSlicer #9635: Sparse infill fragmented path

kisslorand, 2025-05-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

doesn't matter

### How to reproduce

1. Insert a cube
2. Modify it to be brick-shaped (ex.: 60x100x30)
3. Slice with any non-intersecting sparse infill
4. Observe sparse infill path fragmentation

### Actual results

OrcaSlicer v2.3.1-dev
[image]

### Expected results

OrcaSlicer v2.3.0
[image]

### Project file & Debug log uploads

[debug_Thu_May_15_15_31_08_10792.log.zip](https://github.com/user-attachments/files/20228203/debug_Thu_May_15_15_31_08_10792.log.zip)

[Cube.zip](https://github.com/user-attachments/files/20228207/Cube.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

This behaviour leads to unnecessary tool travel, retraction and Z hop, prolonging print time.

kisslorand, 2025-05-20:
For an easier check it is enough to slice the YOLO flow calibration and watch how erratic is the sparse infill.

kisslorand, 2025-05-20:
I think I found the culprit. By reversing PR #9407 the issue goes away.
@elektrohenk can you please take a look? Thank you.

elektrohenk, 2025-05-21:
Hi @kisslorand, [This commit results in the described behavior](https://github.com/SoftFever/OrcaSlicer/pull/9407/commits/6dd71f2c0792c4161cb484680f367da673cf7109). OrcaSlicer is not very smart in ordering the infill paths efficiently. For (aligned) rectilinear there is a special implementation. During implementation this behavior was observed and enforced for solid infill. The special implementation however does not respect the infill anchor parameters. It will connect the lines if the parameter "Maximum length of the infill anchor" is higher than 0.05 mm and none at all if lower. You will see the same "fragmented" infill line order if the lines are not connected. Although the behavior/result is different, it obeys the print settings. In order to keep the print time low, would it be acceptable if the special implementation is enforced when "Maximum length of the infill anchor" is equal or higher than the infill line distance?

kisslorand, 2025-05-21:
@elektrohenk
> [This commit results in the described behavior](https://github.com/SoftFever/OrcaSlicer/pull/9407/commits/6dd71f2c0792c4161cb484680f367da673cf7109).

That commit is responsible for the rectilinear infill type only but the issue is persistent through all the infill type that user straight lines (ex. gyroid is OK).

> For (aligned) rectilinear there is a special implementation that results in a more efficient infill path ordering.

Aligned rectilinear infill is borked also.

> In order to keep the print time low, would it be acceptable if the special implementation is enforced when "Maximum length of the infill anchor" is equal or higher than the infill line distance?

The print time increase is the result of the issue, not the issue itself. I think everybody would agree that a non-erratic infill path is expected regardless of the settings for the infill anchor.

One pattern I observed though is that at the breakage of the infill path the infill anchor goes the other way, towards the already laid down infill lines, instead of its natural course to continue the infill path. Maybe this helps fixing the problem.

elektrohenk, 2025-05-21:
@kisslorand 

> The print time increase is the result of the issue, not the issue itself. I think everybody would agree that a non-erratic infill path is expected regardless of the settings for the infill anchor (as it was prior to PR [#9407](https://github.com/SoftFever/OrcaSlicer/pull/9407)).

Show me an example where the print time increased due to #9407 except for rectilinear and aligned rectilinear.

kisslorand, 2025-05-21:
[image]

[image]

[image]

[image]

etc, etc

elektrohenk, 2025-05-21:
Have you done the same with 2.3.0? I'm not able to find different sparse infill path / printing times for sparse infill patterns other than rectilinear and aligned rectilinear between nightly and 2.3.0 using your Cube project.

[image]

kisslorand, 2025-05-21:
I just literally finished editing my previous message while you posted the above.

> Later edit: I just checked the above examples and sadly they are broken in previous Orca versions also. So it seems PR [#9407](https://github.com/SoftFever/OrcaSlicer/pull/9407) didn't affected others than "Rectilinear" and "Aligned Rectilinear".

elektrohenk, 2025-05-21:
Ok, good to agree on that. Now back to possible solutions.

The best solution would be to improve the path optimization. It seems that `Fill::connect_infill()` is inferior to `ShortestPath::chain_polylines()`. Suspicious: [`//FIXME improve the Traveling Salesman problem with 2-opt and 3-opt local optimization.`](https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/Fill/FillBase.cpp#L1679) I will quickly look into optimizing the infill path using `ShortestPath::chain_polylines()` before calling `Fill::connect_infill()`.

Enforcing the "old" method for "Rectilinear" and "Aligned Rectilinear" when "Maximum length of the infill anchor" is long enough seems to be a need quick fix to me, already successfully tested this solution locally.

kisslorand, 2025-05-21:
While you're at it, it would be just great if you could make all the infill paths to be optimal (shortest path, shortest time).

elektrohenk, 2025-05-21:
Update: Ignore what I said about `ShortestPath::chain_polylines()` vs `Fill::connect_infill()`, both give similar results. [`FillRectilinear::traverse_graph_generate_polylines()` makes the difference, it takes advantage from the properties of the simplified optimization problem for the "Rectilinear" infill pattern where all infill lines are always in the same direction](https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/Fill/FillRectilinear.cpp#L1375-L1379)

elektrohenk, 2025-05-23:
Optimizing the path generation in `Fill::connect_infill()` will take a while. Therefore I will first improve the path generation for "Rectilinear" and "Aligned Rectilinear" when "Maximum length of the infill anchor" is long enough.

closed 2025-09-14 by SoftFever

kisslorand, 2025-09-14:
Yes, it is fixed. Sorry for not letting you know earlier, this whole short segment thing kept me too busy...