OrcaSlicer/OrcaSlicer #9636: OpenCV 4.10 incompatibility

3vi1, 2025-05-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

git

### Operating System (OS)

Linux

### OS Version

Ubuntu 25.10 Questing Quokka

### Additional system information

CPU: AMD Ryzen 9 3900X (24) @ 3.800GHz
Memory:  128GB

### Printer

n/a

### How to reproduce

Compilation on Linux appears to be broken for me due possibly to updates to OpenCV.

I've been successfully compiling the latest git from source for about 6 months, but it's been about 3 weeks since I had time to do so.  Below is the error I receive:

### Actual results

```
[1/11] Performing patch step for 'dep_OpenCV'
FAILED: dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch /home/evil/src/OrcaSlicer/deps/build/dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch 
cd /home/evil/src/OrcaSlicer/deps/build/dep_OpenCV-prefix/src/dep_OpenCV && git apply --directory deps/build/dep_OpenCV-prefix/src/dep_OpenCV --verbose --ignore-space-change --whitespace=fix /home/evil/src/OrcaSlicer/deps/OpenCV/0001-vs2022.patch /home/evil/src/OrcaSlicer/deps/OpenCV/0002-clang19-macos.patch && /usr/bin/cmake -E touch /home/evil/src/OrcaSlicer/deps/build/dep_OpenCV-prefix/src/dep_OpenCV-stamp/dep_OpenCV-patch
Checking patch deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in...
error: while searching for:
        set(OpenCV_RUNTIME vc14) # selecting previous compatible runtime version
      endif()
    endif()
  elseif(MSVC_VERSION MATCHES "^193[0-9]$")
    set(OpenCV_RUNTIME vc17)
    check_one_config(has_VS2022)
    if(NOT has_VS2022)

error: patch failed: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in:137
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in: patch does not apply
Checking patch deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake...
error: while searching for:
    set(OpenCV_RUNTIME vc15)
  elseif(MSVC_VERSION MATCHES "^192[0-9]$")
    set(OpenCV_RUNTIME vc16)
  elseif(MSVC_VERSION MATCHES "^193[0-9]$")
    set(OpenCV_RUNTIME vc17)
  else()
    message(WARNING "OpenCV does not recognize MSVC_VERSION \"${MSVC_VERSION}\". Cannot set OpenCV_RUNTIME")

error: patch failed: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake:176
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake: patch does not apply
```

### Expected results

The patch should apply successfully and compilation complete... but it doesn't.  

The problem appears to be related to the application of 0001-vs2022.patch.  Looking into the autogenerated cmake.in file, I see the following in the MSVC section:

```
  elseif(MSVC_VERSION MATCHES "^191[0-9]$")
    set(OpenCV_RUNTIME vc15)
    check_one_config(has_VS2017)
    if(NOT has_VS2017)
      set(OpenCV_RUNTIME vc14) # selecting previous compatible runtime version
    endif()
  elseif(MSVC_VERSION MATCHES "^192[0-9]$")
    set(OpenCV_RUNTIME vc16)
    check_one_config(has_VS2019)
    if(NOT has_VS2019)
      set(OpenCV_RUNTIME vc15) # selecting previous compatible runtime version
      check_one_config(has_VS2017)
      if(NOT has_VS2017)
        set(OpenCV_RUNTIME vc14) # selecting previous compatible runtime version
      endif()
    endif()
  elseif(MSVC_VERSION MATCHES "^19[34][0-9]$")
    set(OpenCV_RUNTIME vc17)
    check_one_config(has_VS2022)
    if(NOT has_VS2022)
     set(OpenCV_RUNTIME vc16)
     check_one_config(has_VS2019)
     if(NOT has_VS2019)
       set(OpenCV_RUNTIME vc15) # selecting previous compatible runtime version
       check_one_config(has_VS2017)
       if(NOT has_VS2017)
         set(OpenCV_RUNTIME vc14) # selecting previous compatible runtime version
       endif()
     endif()
    endif()
  endif()
```

As can be seen above, the MSVC 193 and greater version lines do not match what the patch is expecting.

### Project file & Debug log uploads

n/a - related to compilation only

### Checklist of files to include

- [x] Log file
- [x] Project file

Noisyfox, 2025-05-15:
Did you try to upgrade OpenCV version in deps?

3vi1, 2025-05-15:
Ah - your other message about the dependencies put me on the right track.  Cleaning them out, I find that everything builds again with no changes.  Thanks a million!

closed 2025-05-15 by 3vi1