OrcaSlicer/OrcaSlicer #9640: Orca Prepare not displaying and crashing during slice/preview on windows/amd cpu/gpu

Ghitim, 2025-05-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Win 11 pro (26100.3915)

### Additional system information

CPU: Ryzen 5950x
Ram: 64gb
GPU: Radeon RX 6900xt

### Printer

Ender 3 Pro

### How to reproduce

1. open Orca
2. open any file/project/calibration
3. view in prepare window (or dont lol)
4. slice or switch to preview

### Actual results

1. open Orca
2. open any file/project/calibration
3. view nothing in the prepare model area
4. try to slice or switch to preview
5. orca crashes

### Expected results

1. open Orca
2. open any file/project/calibration
3. view in prepare window
4. slice or switch to preview
5. print(upload) to octoprint/export gcode

### Project file & Debug log uploads

[project.zip](https://github.com/user-attachments/files/20232653/project.zip)

[log.zip](https://github.com/user-attachments/files/20232669/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

the project file uploaded is just an example, it'll happen on any project or stl import, or calibration file created. etc. etc.

I've tried reinstalling orca, I've tried with wiping the orca directory and the appdata orca folders and reinstall with no change (other than having to run through setup again)

i didn't update my video card drivers, they've been the same for a month with no issues and there's nothing more up to date.
this just suddenly started happening upon one load up, only thing i did around that time was swap around my physical monitors.

i looked around and found similar issues here for linux computers with NVIDIA gpu's, but nothing for windows or amd

related? changing from light to dark mode or vice versa also crashes orca (but the change is effected next load)

Ghitim, 2025-05-15:
did some more messing around with my physical connections and now everything works as expected again and i cant force it to break.... idk man. computers lol

closed 2025-05-15 by Ghitim

Enohriel, 2025-05-22:
System : 9800X3D / 6900XT

I do have the same issue.
Sometimes it works, sometime it doesn't.
It does not prevent me from slicing, but I can't see Preview, and Thumbnails are broken.

[image]

Going back on Prepare/Preview tab :

[image]