OrcaSlicer/OrcaSlicer #9645: admesh potential licensing issue and lost provenance due to significant changes to the alleged upstream admesh

Centuriondan, 2025-05-16:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

none - attempting to package 2.3.0 as a debian package.

### Operating System (OS)

Linux

### OS Version

Devuan Excalibur (equivalent to the Debian Trixie (next release) (but irrelevant)

### Additional system information

Irrelevant to this bug.

### Printer

irrelevant to this bug

### How to reproduce

"Read the Source, Luke"
Comparison between src/admesh (inherited from Bambu Studio) and https://github.com/admesh/admesh 

Executive Summary of this issue:

The "Additional Authors" noted in the head comment in the files are not actually known or knowable as the AUTHORS file is not included in this rendition of admesh.  There is no provenance for the authorship of the C++ code which carries the same head comments as the original stl.h file - this should be a concern with respect to the origins of this new code (from Bambu Studio).

Whilst this is a low risk of becoming an issue, it would be nice to have it tidied up - which probably means asking Bambu Labs to amend the file headers and restore the Authors files from the Upstream project.

Background notes:
Only the stl.h file is preserved from the upstream project, and all the new c++ code and headers are new along with the cmake config.

The original as best as I can find is licensed as GPL v2 (or later)  which is fine (there is some discussion as to whether AGPL is a really a later version or a different license - so it may be worth noting this in the Licensing section of OrcaSlicer.

### Actual results

N/A

### Expected results

Correct attributions and license information shared.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-05-16:
I think this was mostly done by Prusa https://github.com/prusa3d/PrusaSlicer/commit/138501872426ae1fc7023478dd1deba96d936ecf#diff-3b4954ee91d3e2ee8f74a7f2cbed2ebdcbaaf6f3450e208100c7a93b594fba91

And we don't have much knowledge on why and what they did to this lib.

c2h5oh, 2025-05-16:
It goes even deeper than that:
- there are OrcaSlicer changes https://github.com/SoftFever/OrcaSlicer/commits/main/src/admesh
- on top of BambuStudio changes https://github.com/bambulab/BambuStudio/commits/master/src/admesh
- on top of PrusaSlicer changes https://github.com/prusa3d/PrusaSlicer/commits/master/bundled_deps/admesh + https://github.com/prusa3d/PrusaSlicer/commits/fd02400245df25f1286059aa642739660df76663/src/admesh
- on top of Slic3r changes https://github.com/slic3r/Slic3r/commits/master/xs/src/admesh
- on top of admesh that is planning to pull some changes at least from slic3r at some point https://github.com/admesh/admesh/issues/21 for almost a decade now

Each layer of changes has it's own unique changes as well as ports or direct copies of changes from one or more of the other layers.

Nobody seem to have bothered to update AUTHORS.