OrcaSlicer/OrcaSlicer #9661: One of 2 identical bridges is directed wrong in the same print.

TLmv, 2025-05-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambulab x1c

### How to reproduce

make bridges, i made sets of 3 ridges of 200mm, outer 2mm, then 4.6mm open, .6mm wall, 4.6 open and wall. after running through the slicer one wall came out bridges over the 10mm width, however, the left wall (same layer, printed after right bridge) was sliced over the 200mm length, making it impossible to print and thus it failed

### Actual results

[image]

[image]

### Expected results

that the bridge will not allow larger than, lets say 20mm. that is probably already way too much, so it's a safe assumption.
if larger it should then choose to alter direction if that gives a shorter bridge.

### Project file & Debug log uploads

[wrongbridgeleftLayer28_PLA_5h38m.gcode.zip](https://github.com/user-attachments/files/20265676/wrongbridgeleftLayer28_PLA_5h38m.gcode.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TLmv, 2025-05-21:
what i also think could be improved is you can see that it first creates an overhang wall before it tries to bridge the gap.
if we are building layer on layer, wouldn't it be more prudent that bridges start on a wall instead of trying to knit them to an overhang?
i think it would improve strength and less chance of hanging bridges.

john-parton, 2025-05-26:
What happens if you set "External Bridge Infill Direction" to 0.000001 (one millionth of a degree)?

In theory that should do nothing, but I found it "worked around" a possibly related issue: https://github.com/SoftFever/OrcaSlicer/issues/9725

TLmv, 2025-05-26:
It then still first creates the overhang walls that weaken any bridge instead of first printing bridges and then see if there are overhangs left to print. 
but it does solve the wrong direction left bridge.
so, cudo's for this work around,

can i save that setting for all project or could that lead to other problems and should i only use it when i encounter wrongly directed bridges?

chris99336, 2025-06-03:
it is worse than what is being shown. Even when there are no walls, it is doing the dumbest thing I have ever seen and drawing the bridge back and forth with nothing to support it.

[image]

TLmv, 2025-06-03:
If that would be true why has my example then 2 directions on the same 
z-level.

Op 03-Jun-25 om 14:44 schreef chris99336:
> That's a really hard print for bridging because of the way bridge layers 
> are calculated. The way the bridge layer works, all of the lines have to 
> go in the same direction if they are on the same z-level.

MakerMatrix, 2025-06-20:
Is this related, or am I confused?

[image]

My expectation was that this first external bridge layer would traverse between the two pillars. This was with "External bridge infill direction" set to 0, so the slicer calculates what to do.  It failed in my view.  But maybe I am confused and need more education about what developers expect from modern printer hardware.  I would not expect that infill to fuse to the inner wall, in mid-air and in fact it did not on two attempts (trying different, even slower, speeds now).

closed 2025-09-28 by github-actions[bot]