OrcaSlicer/OrcaSlicer #9663: Supports inside support blocker

tlhintoq, 2025-05-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1 dev build 1a5b238

### Operating System (OS)

Windows

### OS Version

10

### Additional system information

11th gen multicore

### Printer

Any

### How to reproduce

Lay out an object and add support blocker cubes and slice it.

### Actual results

Supports inside blockers

### Expected results

No supports inside support blockers

[image]

### Project file & Debug log uploads

[WhySupportsOnBack.zip](https://github.com/user-attachments/files/20271880/WhySupportsOnBack.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

See project attached.

LE1ED4523, 2025-12-14:
support block can't block support 
2.3.0 have same issue

Azivast, 2026-02-05:
Still present in the 2.3.1 release

tlhintoq, 2026-10-30:
<bot nudge>