OrcaSlicer/OrcaSlicer #9665: V2.3.0 not working on Zorin OS 17.3

jinhooo, 2025-05-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Zorin os 17.3

### Additional system information

`Kernel: 6.8.0-59-generic x86_64
Desktop: GNOME 43.9 tk: GTK 3.24.33 wm: Openbox dm: GDM3 42.0
Distro: Zorin OS 17.3 base: Ubuntu 22.04 LTS Jammy
Mobo: ASRock model: B550M Steel Legend
model: AMD Ryzen 7 5700X
NVIDIA GA106 [GeForce RTX 3060 Lite Hash Rate] vendor: ZOTAC`

### Printer

qidi plus 4

### How to reproduce

1. Download the appimage from GitHub 
2. start and configure OrcaSlicer
3. connect printer to OrcaSlicer
4. slice something send it to print 
5. then try printing something else 
After that it will not work anymore and will crash every time i click on OrcaSlicer
At the time I was just calibrating my filament so I have no project files. I did the temperature tower and it said there were some errors and it couldn't solve it because i wasn't on windows. I just sliced it and it printed just fine.

### Actual results

It just crashes and when i try open it from the terminal it just crashes with a bunch of errors like segmentation fault, no such file in directory, webkit bug, ...
`me@me-desktop:~/Desktop/qidi$ ./OrcaSlicer_Linux_AppImage_V2.3.0.AppImage
[2025-05-18 15:40:23.932190] [0x00007c1522d044c0] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.670: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.670: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15660): GLib-GObject-WARNING **: 15:40:24.670: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.670: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.671: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.671: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15660): GLib-GObject-WARNING **: 15:40:24.671: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.671: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.672: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:24.672: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:15660): GLib-GObject-WARNING **: 15:40:24.672: invalid cast from 'wxPizza' to 'GtkCellLayout'
[... 293 lines cut ...]

(orca-slicer:15660): Gtk-CRITICAL **: 15:40:25.729: gtk_widget_set_size_request: assertion 'width >= -1' failed
No such file or directory
Segmentation fault
me@me-desktop:~/Desktop/qidi$ ERROR: WebKit encountered an internal error. This is a WebKit bug.
./Source/WebKit/WebProcess/Network/WebLoaderStrategy.cpp(577) : internallyFailedLoadTimerFired`

### Expected results

When you download it at first it should work fine, but when you connect your printer to OrcaSlicer and slice/print your thing it will not work anymore and it crashes every time. In fact you will not be able to slice and anything it will just crash immediately. The same happened to me on manjaro. 
I also have QIDI studio and it works just fine no issues like that.

### Project file & Debug log uploads

[debug_Sun_May_18_15_46_38_16820.log.zip](https://github.com/user-attachments/files/20273352/debug_Sun_May_18_15_46_38_16820.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

I also tried changing to X11 like someone [did](https://www.reddit.com/r/OrcaSlicer/comments/1gbfbpg/suddenly_orcaslicer_started_to_fail_on_linux/) but it didn't work 
someone in the same thread said that if you disabled your internet connection OrcaSlicer will work, and it did work but I am unable to send prints via internet (which is normal)

thisnyjared, 2025-05-19:
Dose orcaslicer work with previous versions like 2.2.0? Or is this issue just about this specific version. It might be helpful to check if you haven't.

jinhooo, 2025-05-19:
it does not work with previous version. it does the same thing.

B20bob, 2025-05-28:
I am having the same issue on Pop!OS 22.04 LTS.

Everything works fine on windows 11 using the portable app.

I originally thought the crash was being caused by my profiles which I imported, but I have narrowed the issue down to being related to the printer connection config. Orca runs fine until I add my printer address and close that dialog. The app then crashes almost instantly, it will crash within seconds of launching every time I try to open it after that until I disable my Ethernet connection and remove the address from the printer connection config. Has anyone found a workaround for this?