OrcaSlicer/OrcaSlicer #9667: Flooded by compiler warnings

rubienr, 2025-05-18:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

As a developer i want to build `OrcaSlicer` from source and test my changes, but i am flooded with a pot of compiler warnings any time i build from source. Hence i cannot easily spot my self made issues (needle in haystack).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Please reduce compiler warnings in a meaningful way and allow others to easier spot their self introduced warnings.

Suggestions:

- silence warnings from 3rd party libraries as we cannot do anything about it
- fix the lowest hanging fruits of compile time warnings, even it its just naive zero-initializing values (the compiler will never the less optimize and remove initialization if really not required) or removing unused variables
- introduce coloured compile output

henrivdr, 2025-05-19:
I like the suggestion about the 3rd party libraries.

I see you already did PRs for a lot of this

rubienr, 2025-05-19:
I would like to help out even more with warnings once this PR is merged.

rubienr, 2025-05-21:
I roughly compared the compile output of this PR vs branch `main` and found that it reduces about `4873` (59%) lines of compile output:

- `main` branch total log lines: `8307`
- this PR branch: `3434`

How did i test:

````bash
rm -drf build deps/build
BuildLinux.sh -d
BuildLinux.sh -sc | tee compile-output.txt
...
cat compile-output.txt | wc -l
```

PR #9619 (2025-06-14): fixes compiler warnings
# Description

Fixes #9667: the lowest hanging fruits of compiler warnings

- no features were introduced/modified
- eliminates several compiler warnings:
  - uninitialized variable
  - unused variable
  - unused return value
  - suggested parentheses
  - suggested curly braces
  - redundant template type
  - operator precedence mismatch (`?` vs `<<`)
- CMake
  - adds `SYSTEM` to `target_include_directories()` and `include_directories()` to suppress warnings from dependencies