OrcaSlicer/OrcaSlicer #9682: Hotkeys not working on Ubuntu 24.04 with 2.3.0 AppImage

kleinwareio, 2025-05-19:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04

### Additional system information

```
$uname -a

Linux me 6.11.0-25-generic #25~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Apr 15 17:20:50 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

$lsb_release -a

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04.2 LTS
Release:	24.04
Codename:	noble

$echo $XDG_SESSION_TYPE
x11

$echo $DESKTOP_SESSION
ubuntu

$ dpkg -l | grep -E 'gtk|qt'
ii  apport-gtk                                    2.28.1-0ubuntu3.5                          all          GTK+ frontend for the apport crash report system
ii  gir1.2-gtk-3.0:amd64                          3.24.41-4ubuntu1.3                         amd64        GTK graphical user interface library -- gir bindings
ii  gir1.2-gtk-4.0:amd64                          4.14.2+ds-1ubuntu2                         amd64        GTK graphical user interface library -- gir bindings
ii  gir1.2-javascriptcoregtk-4.1:amd64            2.48.1-0ubuntu0.24.04.1                    amd64        JavaScript engine library from WebKitGTK - GObject introspection data
ii  gir1.2-javascriptcoregtk-6.0:amd64            2.48.1-0ubuntu0.24.04.1                    amd64        JavaScript engine library from WebKitGTK - GObject introspection data
ii  gir1.2-nma4-1.0:amd64                         1.10.6-3build2                             amd64        GObject introspection data for libnma-gtk4
ii  gtk-update-icon-cache                         3.24.41-4ubuntu1.3                         amd64        icon theme caching utility
ii  gtk2-engines-murrine:amd64                    0.98.2-4                                   amd64        cairo-based gtk+-2.0 theme engine
ii  gtk2-engines-pixbuf:amd64                     2.24.33-4ubuntu1.1                         amd64        pixbuf-based theme for GTK 2
ii  ibus-gtk:amd64                                1.5.29-2                                   amd64        Intelligent Input Bus - GTK2 support
ii  ibus-gtk3:amd64                               1.5.29-2                                   amd64        Intelligent Input Bus - GTK3 support
ii  ibus-gtk4:amd64                               1.5.29-2                                   amd64        Intelligent Input Bus - GTK4 support
ii  libcanberra-gtk3-0t64:amd64                   0.30-10ubuntu10                            amd64        GTK+ 3.0 helper for playing widget event sounds with libcanberra
ii  libcanberra-gtk3-module:amd64                 0.30-10ubuntu10                            amd64        translates GTK3 widgets signals to event sounds
ii  libcolord-gtk4-1t64:amd64                     0.3.1-1build2                              amd64        GTK4 convenience library for interacting with colord
ii  libdbusmenu-gtk3-4:amd64                      18.10.20180917~bzr492+repack1-3.1ubuntu5   amd64        library for passing menus over DBus - GTK-3+ version
ii  libdecor-0-plugin-1-gtk:amd64                 0.2.2-1build2                              amd64        libdecor decoration plugin using GTK
ii  libdecor-0-plugin-1-gtk:i386                  0.2.2-1build2                              i386         libdecor decoration plugin using GTK
ii  libgtk-3-0t64:amd64                           3.24.41-4ubuntu1.3                         amd64        GTK graphical user interface library
ii  libgtk-3-0t64:i386                            3.24.41-4ubuntu1.3                         i386         GTK graphical user interface library
ii  libgtk-3-bin                                  3.24.41-4ubuntu1.3                         amd64        programs for the GTK graphical user interface library
ii  libgtk-3-common                               3.24.41-4ubuntu1.3                         all          common files for the GTK graphical user interface library
ii  libgtk-4-1:amd64                              4.14.2+ds-1ubuntu2                         amd64        GTK graphical user interface library
ii  libgtk-4-bin                                  4.14.2+ds-1ubuntu2                         amd64        programs for the GTK graphical user interface library
ii  libgtk-4-common                               4.14.2+ds-1ubuntu2                         all          common files for the GTK graphical user interface library
ii  libgtk-4-media-gstreamer                      4.14.2+ds-1ubuntu2                         amd64        GStreamer media backend for the GTK graphical user interface library
ii  libgtk2.0-0t64:amd64                          2.24.33-4ubuntu1.1                         amd64        GTK graphical user interface library - old version
ii  libgtk2.0-bin                                 2.24.33-4ubuntu1.1                         amd64        programs for the GTK graphical user interface library
ii  libgtk2.0-common                              2.24.33-4ubuntu1.1                         all          common files for the GTK graphical user interface library
ii  libgtk3-perl                                  0.038-3                                    all          Perl bindings for the GTK+ graphical user interface library
ii  libgtkmm-3.0-1t64:amd64                       3.24.9-1                                   amd64        C++ wrappers for GTK+ (shared libraries)
ii  libgtkmm-4.0-0:amd64                          4.10.0-4build3                             amd64        C++ wrappers for GTK4 (shared libraries)
[... 56 lines cut ...]
- Hotkeys when focused in the "side bar", such as `?` or `1/2/3/...` to change extruders
    - These bring up a search box looking thing instead

### Actual results

Some hot keys are ignored or bring up a search box looking thing.

[image]

### Expected results

The hot key command is executed

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I didn't have this problem with `2.3.0_beta` running on Windows, so I assume its a Linux thing...

hubiscode, 2025-05-21:
I have the same issue. I believe it's a keyboard focus problem. If there is an info pop-up in the bottom-right corner, closing it will make the shortcuts work for me. Also, in the workspace twice (once after the pop-up appears) will make the shortcuts work again.

kleinwareio, 2025-05-21:
Thanks, I can also confirm that closing the "Object info popup" in the lower right corner then allows the number hotkeys to work. Unfortunately, as soon as I click on another object the popup comes back, so this isn't really a workaround, but its probably a good hint about where the issue is.

kleinwareio, 2025-11-05:
Still an issue

Felix14-v2, 2026-09-24:
Could you try the [latest Nightly](https://github.com/OrcaSlicer/OrcaSlicer/releases/tag/nightly-builds)? There were some changes in shortcut handling.

kleinware, 2026-09-25:
(edit: I tested `OrcaSlicer_Linux_AppImage_Ubuntu2404_nightly.AppImage` downloaded 2026-09-24 Z ~16:00 on Fedora 43)

Thank you!

Observations:
- the "search box" thing popping up seems to be fixed
- Not having anything selected and pressing 'a' arranges as expected
- From nothing selected (click on empty spot on build plate)
    - click an object ("Object name" card becomes visible)
    - now hot keys are blocked / not working
    - Click the "X" to close the "object name" card
    - hot keys work again
- From nothing selected (click on empty spot on build plate)
    - click an object
    - click another object ("Object name" card still visible)
    - now hot keys work
- From nothing selected (click on empty spot on build plate)
    - click an object
    - click same object again
    - now hot keys work
- From nothing selected (click on empty spot on build plate)
    - click an object
    - click another object (get hot keys working)
    - press CTRL+X (object is cut)
    - press CTRL+V and it doesn't work (object is not pasted)
    - Click on plate (no object) then press CTRL+V and object is pasted
    - Press CTRL+V again and nothing happens
- With having object in copy buffer
    - Make new plate
    - click plate
    - Ctrl+V pastes object
    - Ctrl+V again does nothing
    - Click plate
    - Ctrl+V does nothing
    - Click plate again
    - Ctrl+V pastes
    - Cycle continues: click plate two times, Ctrl+V, click plate two times, Ctrl+V, ...

So, overall a big improvement, thanks! It's much more usable now knowing that something about the first focus seems to be blocking the hot keys makes it easier to work around. So, if I want to change an object to filament 2 then I just have to click it, click it again, and then hit 2. This "click twice" also seems to apply to the build plate, as seen in the "trying to paste a bunch of copies of an object with CTRL+V" scenario at the end of the above list.

Thanks again for working on this!