OrcaSlicer/OrcaSlicer #9688: Support Interface Filament - Use Only For Big Support Layers

JensWurm, 2025-05-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

When using a special filament as the support interface, i am frustrated that this only really works well if there are only few flat interfaces. It does not work well if there is a great number of support points, due to the excessive number of filament changes.

A model may have a few flat surfaces that would benefit greatly from

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be great to have a more fine-grained control over the support interface filament setting. Perhaps using area size limits, or filament usage limits. 

Something like, "use the support interface filament only if the area to be supported is at least 5 square millimeters" or "use the special support interface filament only if this support layer overall is at least 0.1g of filament" - whatever metric is available in the slicer.

Alternative: Manually paint on the model which support interface filament is to be used for maximal customization

That way one could avoid needless filament changes for very small supports

Ideally one would be able to also have custom interface settings then. Custom z offset and interface spacing for the special interface filament, normal values for the default filament.

### Additional context

[image]

226 filament changes - most of those supports really didn't need the special interface filament. There were just a handful of support interfaces that really needed it.

igiannakas, 2025-05-21:
That is a pretty cool idea! Something like setting a minimum threshold area size to use support material vs regular material for the interface. The only challenge would be for slanted surfaces, where each layer has only a small threshold but over multiple layers they are supported over a larger surface.

JensWurm, 2025-05-21:
Yeah, being able to skip the support material for those is one of the things i'd like to be able to do. 
IMHO slanted surfaces need the support material far less urgently than flat ones as they are partially supported by the model already.