OrcaSlicer/OrcaSlicer #9691: Warning dialog about moving/jogging not homed printer

victornpb, 2025-05-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

[image]

### Problem

1. Motion starts before warning. A jog command executes, then the “please home” dialog appears. If the head is already outside limits, damage can occur before the user can react.

2. A few times I knew the printer wasn't hommed but I needed to Move it, and I accidentally pressed the "Home" button on the dialog since you have to dismiss it on every click, Causing the printer to crash into object. When you need to move an un-homed machine (e.g. to clear filament), a slip-click on Home drives the nozzle into whatever’s on the bed.

3. No temporary dismissal. The dialog re-appears on every jog, forcing extra clicks and increasing the odds of #2.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

### Suggestion

This dialog should instead be:

[image]

```
Printer not homed - movement may exceed limits and damage hardware.

Please Home all axes first clicking 🏠 button (make sure the bed is clear).

[x] Don’t show this warning again for 1 minute.

[ Cancel ] [ Move anyway ]
```

Including a Home button in a dialog triggered by a Move command violates the UX principle of contextual relevance: users expect only actions directly related to the task they initiated. Mixing an unrelated, high-risk option increases cognitive load and the chance of a catastrophic mis-click (Fitts’s Law + error-prevention heuristics). 

So I suggest no explicit Home button. The user initiated a move, so the dialog should only let them cancel or proceed; homing should be a suggestion, and not a primary action.

I kept the message purposely short and and with very high signal to noise ratio, working with UX for a while, dialogs that are direct and short have a much betters success than dialogs that use more verbose english, (e.g.: Delete file? VS Are you sure you want to proceed deleting the file?).

### Describe alternatives you've considered

henrivdr, 2025-05-21:
Are you sure this is for Orca Skicer?

victornpb, 2025-05-23:
> Are you sure this is for Orca Skicer?

Yes, the screenshot is from bambustudio but orcaslicer dialog is identical. I think it's something I wanna see on orcaslocer and maybe it will make to upstream

henrivdr, 2025-05-23:
I think that is the printer's web interface , not the slicer GUI.
Do you see it in the Device tab?
What does it look like via a Web Browser?