OrcaSlicer/OrcaSlicer #9692: Add multiple prime towers/ additional settings for prime tower issues for multiple filament types

s3anami, 2025-05-20:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

The current prime tower settings do not work effectively with dissimilar filament types (example: pla/petg).
The tower usually breaks at some point in its structure (base or middle)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

The newer Bambu Studio settings for prime towers seem to work better with dissimilar filaments like PETG and PLA, so having similar setup to that would reduce the issue. Or implementing multiple separate prime towers. Or another solution someone smarter has thought of

### Describe alternatives you've considered

I have just stopped attempting to use dissimilar filaments for supports as it wont work with prime towers
Switching to other slicer for non tool changer machines

GreavesDesignerOne, 2025-09-02:
Discussion #5348 has brought up this issue, and Issue #7644 (stale, no activity) requests a feature to solve this issue as well. There may be others, I found those during a quick search. Both comments suggest either separate Prime Towers for each filament type, or having the Prime Tower use different materials for the inner and outer walls.

Printing with PLA as the support material for PETG seems to have become pretty common for machines with dual extruders as it allows more complex geometries using affordable filaments, reduces post-processing time and improves part quality. I have had several large prints fail as a result of the prime tower delaminating during the print due to material incompatibility. A solution to this issue would be greatly appreciated!

Baggles5, 2025-12-05:
I also have this issue.

banffman, 2026-01-30:
This would be very helpful.

Hisenmisen, 2026-08-17:
This would benefit me aswell if the prime tower deosnt print TPU and PLA on top of eachother. It actualy caused a bit of contamination in my TPU which led to separated layers in the actual print.