OrcaSlicer/OrcaSlicer #9699: Flushing Volume does not work for Creality Printers

efras92, 2025-05-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Printer

Creality K2 Plus/ Creality Hi/ All Creality Printers

### How to reproduce

Slice multicolor model with Creality Printer Selected. Color scheme - Filament does not include flush amount.

Slicing with any multiplier changes nothing. 

[image]

### Actual results

No matter the value of the  flush value multiplier printer does not purge any different leading to bleeding colors and unusable prints.

I tested anything from 0 to 3.0 multipliers and the filament amount and print time are not affected. Tested on 4 Prints, each with different flushing volume set and the results are identical and not usable

### Expected results

Increasing Flushing Volume should increase the actual flushing volume? 

### Project file & Debug log uploads

[debug_Wed_May_21_10_58_27_62140.log.zip](https://github.com/user-attachments/files/20377547/debug_Wed_May_21_10_58_27_62140.log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Zionmatrix2025, 2025-07-16:
so purges are different then the prime tower. you can purge to the prime tower, but gives the super big tower.
otherwise the purge setting all goes into printer poop. high number (I think above .8 - 500mm3) it should dump the poop twice between colors.

JAFC-BandO, 2025-09-18:
I'm having simlar issue on K2 Combo its not reacting to the flushing volumes at all ? It always just poops out a lot, no matter if it's 0 or 1000. Can somebody please help? 

I'm using Creality Print 6.2... :(

Stevetm2, 2026-03-06:
@Matigon16 
if you add this Post Processing Script to Orca, do you then get any different results?  I noticed that Creality Print was checking for a few gcode file "markers" so, in the script, I swap some strings out of the Orca generated files, in the attempt to fix this.  I havent had time to do any color prints recently, so I dont know if it heps with the prints using the 1.1.4.11/1.1.4.8 K2 Plus firmware/s.  It should do the same for K1+CFS or K2+CFS btw...

https://github.com/Stevetm2/K2_Custom_Macros/tree/main/OrcaSupportFiles/PostProcessingScripts/OrcaToCrealityFilamentSelect