OrcaSlicer/OrcaSlicer #9701: Set support type per modifier object (allow modifier objects to specify different support)

alexfornuto, 2025-05-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

This issue is to bring light to #3658, which was closed by bots before it could be addressed. I'll not waste time repeating the contents here, I'm just hoping that a maintainer will see this and re-open #3658 so it can get proper attention.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

#3658 gets re-opened and tagged so that bots don't close it again.

Acelya-9028, 2025-08-31:
That would be really awesome :D

alexfornuto, 2025-10-22:
Just bumping so that the bots don't auto-close this as they did the original.

ScottHagie, 2026-03-17:
For multimaterial printing using different kinds of supports in different places, this is required.

Mauker1, 2026-05-07:
This is a must, especially now that we have easier access to tool changers! I tried to set different support interfaces for different materials in my model, but there was no way to do it :(