OrcaSlicer/OrcaSlicer #9702: Using the "Brim Ears" toolbar helper can break Brim options for an object

csstup, 2025-05-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

windows 10/11

### Printer

Any

### How to reproduce

1. New project
2. Add object (in this case, an STL)
3. Use the "Brim Ears" toolbar feature, press the "set brim type of this object to painted" option to allow generating brim ears.
4. From this point forward, the object can no longer have brim options set other than the painted brim ears from the tool.
4a. The _Brim type_ doesn't change to "Painted", its left unchanged, which is odd.
4b. Even if you remove all the brim points created in the Brim Ears tool, you can't set a brim to something else like "Outer brim only".    
4c. If you force set the _Brim type_ to something else, its ignored.  Set it back to None, still ignored.
5. The only fix appears to be to delete the object and import it again.

This is duplicatable in 2.3.0 release and in the current 2.3.1 nightly build as of 5/22/2025.

### Actual results

Dialog in question, use the "set brim type of this object to painted"
[image]

"After trying to clear all previous brims and then create a new outer only (for example), no brim can be created
[image]

You have to delete the object and reimport it and start over if you want to affect brims.

### Expected results

1. I think the "set brim type to painted" should actually change the brim type of the object to "painted".  As of now it doesn't change the value.
2. When overriding this with a new brim type it should remove the "painted" application and do what the user selected.

### Project file & Debug log uploads

Any project, doesn't matter.
No log needed, didn't crash.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-05-23:
Switch to object mode. This brim type is set only to that object, as the text suggested.

csstup, 2025-05-23:
Thanks for the followup.    Theres no text that says its setting it on the object, which led to my confusion.  

[image]

Also confusing, in Object mode, there is no scrollbar on the bottom indicating that there are "tabs" to the left or right of the current one.  I had the overall panel pretty narrow and had no idea "Other" was hiding to the right. 

Global, has scrollbar if any of the tabs are hidden
[image]

Object, no scrollbar:
[image]

If you make the window smaller than this, the scrollbar appears.    But I had it right at this size where Other is hidden but no scrollbar shows up.

If thats a different bug I can make a report for that if it would help.

closed 2025-08-30 by github-actions[bot]