OrcaSlicer/OrcaSlicer #9704: bridge over supports?

fauks, 2025-05-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

A way to make the layer above supports be bridge instead of overhang. 

Superslicer does this but Orcaslicer does not. 

OrcasSlicer:
[image]
SuperSlicer:
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

An option to treat solid infill over supports as bridges. perimeters can still be treated as overhang.

fritzw, 2025-05-23:
You can get the same behavior as SuperSlicer if you uncheck the "Extra perimeters on overhangs" setting. This has annoyed me several times, but I'm not sure if this should be considered a bug. This Prusa forum thread gives some compelling arguments in favor of this feature: https://forum.prusa3d.com/forum/prusaslicer/what-is-the-extra-perimeters-on-overhangs-setting-actually-good-for/

fauks, 2025-05-24:
Wow! I thought I had tried every setting trying to get this behavior but I guess I skipped "extra perimeters on overhangs" for whatever reason. 

I didn't realize it starts in the middle and I can see how that could be useful but I tend to get better results above supports if they are bridged. Maybe I need to tune my support Z offset a bit but I always get curling so the supports can't do their job. 

I appreciate your response, being able to deselect that setting when needed is probably fine. 

Thank you!

closed 2025-05-24 by fauks