OrcaSlicer/OrcaSlicer #9723: [Filament] Add Transparency to Colourpicker

DerNinsaa, 2025-05-25:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Yes. When setting up MMU filament profiles, it's currently not possible to define translucent or transparent filament colors, since the color picker only allows opaque (solid) colors. This makes it harder to visually distinguish translucent filaments in the preview or manage them accurately. Additionally, the purge values in the MMU context may be inaccurate, as the visual representation does not reflect the real-world material properties (e.g. clear-to-dark transitions requiring more purge than light-to-light).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I’d like the filament color picker to support transparency – for example, by allowing RGBA input, an additional opacity slider, or support for hex codes with alpha (like #ffffff80). This would make it easier to represent transparent or semi-transparent filaments in MMU workflows.

### Describe alternatives you've considered

I’ve tried using very light grays to simulate transparent filaments, but this doesn't capture the visual difference between a white filament and a clear one. Manually editing config files to include transparency also doesn't work, as the current UI and rendering don't support alpha values.

### Additional context

None, Thank you so much for your Work on OrcaSlicer!

pi-squared-studio, 2025-05-25:
If this goes well, then we can create the opportunity to show artistic plastics such as PLA Silk, Glow, Glitters, Rainbow, Bi-Tricolor... For the latter, you can provide the Phase parameter when viewing the model to set the initial angle or color. These parameters will help determine the final appearance of the model with alternating colors. With these settings, Orca will be the most representative of all slicers.

slothy89, 2025-06-03:
I support this.
Ability to represent Gradients and co-extruded filaments would be nice too. Probably more-so than transparency for me.

rkarlsba, 2025-09-29:
I've wanted this for a long time. Also, please add particles so that such things as marble-like filament can actually look like it.

yw4z, 2026-10-17:
i assume this can be closed. https://github.com/OrcaSlicer/OrcaSlicer/pull/16197 merged already
also a PR for transparency support on AMS

closed 2026-10-23 by ianalexis