OrcaSlicer/OrcaSlicer #9725: Extremely long bridges where shorter bridges should be possible

john-parton, 2025-05-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 25.04

### Printer

Ender 3 S1-Pro

### How to reproduce

I've attached the project file where the behavior can be seen.

### Actual results

Sometimes bridges are extremely long. The model has two roughly parallel sides and I would expect the bridges to be as short as possible, going roughly 90 degrees to the walls.

### Expected results

Here's an example where extremely long bridges are generated, where shorter bridges should be possible

[image]

### Project file & Debug log uploads

[unoptimized-bridges.zip](https://github.com/user-attachments/files/20434528/unoptimized-bridges.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

john-parton, 2025-05-26:
This is actually half of a project. The other half is basically identical, and it generates the bridges in the expected way.

[image]

john-parton, 2025-05-26:
This might be a user error on my part, but because it generated it correctly for the other half of the project, it seemed like it was a bug. 

Possibly related:

https://github.com/SoftFever/OrcaSlicer/issues/8276
https://github.com/SoftFever/OrcaSlicer/pull/8263
https://github.com/SoftFever/OrcaSlicer/issues/9599

Possible duplicate (I looked before I submitted, I just didn't see it)

https://github.com/SoftFever/OrcaSlicer/issues/9661

john-parton, 2025-05-26:
OK, so something interesting. If I change "External Bridge Infill Direction" to `0.000001` (one **millionth** of a degree), it actually generates as expected. 

Ordinarily I would say it's up to the user (me) to understand the specific options and apply them appropriately, but it seems like there might be some sort of floating point arithmetic error or something where it might be appropriate to apply some sort of numerical fix.

mbravorus, 2025-06-13:
+1 experiencing exactly the same problem

closed 2025-09-19 by github-actions[bot]