OrcaSlicer/OrcaSlicer #9732: OrcaSlicer V2.3.0+ hangs on splash screen when AppData\Roaming is redirected to a network location (Windows 11 Pro)

pcdrivers, 2025-05-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Additional system information

I'm experiencing an issue with OrcaSlicer starting from version 2.3.0 (also tested on the nightly 2.3.1) on a Windows machine that's part of a domain. In this setup, the user profile's AppData\Roaming folder is redirected to a network share (e.g., \SERVER\Users\Username\AppData\Roaming).

The issue:

OrcaSlicer launches correctly the first time.

On subsequent launches, it hangs indefinitely at the splash screen (logo).

If I delete the OrcaSlicer folder from Roaming, it launches again once — but the problem returns after restarting.

Additional info:

Logging in as a domain administrator works fine.

I’ve given full permissions to the user and even tested with domain admin rights — no change.

File access from the user works — I can read/write/delete files in the network Roaming path.

It seems the problem occurs when OrcaSlicer tries to use a remote path for configuration, rather than a local one like C:\Users\Username\AppData\Roaming....

This behavior did not happen with previous versions of OrcaSlicer — only with 2.3.0 and newer.

Any suggestions or workarounds would be appreciated.

### Printer

Anyone

### How to reproduce

I have tested on different computers running Windows 11 Pro, all of them joined to a domain and using roaming profiles (not in a workgroup).
When I delete the folder located at %appdata%\OrcaSlicer, the program launches correctly the first time.
However, after closing the program and trying to open it again, it fails to start.

### Actual results

OrcaSlicer cannot be opened unless I delete its folder from AppData and then log in to Bambu Lab to re-download my profiles.
I have also tried removing only the configuration files, but the problem persists.

### Expected results

.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-05-28:
Could you run ProcMon and see if it stuck as certain IO operation? Meanwhile I'll try reproduce this locally.

pcdrivers, 2025-05-28:
Hello Noisy,

I can send you the Procmon log. There are no obvious errors, just some missing files (not found). It looks like OrcaSlicer might not be loading the correct folder:
"\\TSSERVER2\Usuarios\christian\AppData\Roaming\OrcaSlicer"

Also, the OrcaSlicer debug log is 0 bytes.

[procmon_log.zip](https://github.com/user-attachments/files/20482235/procmon_log.zip)

pcdrivers, 2025-05-28:
> Did you have bambulab network plugin enabled?

Yes, enabled and installed. When i delete the folder Orcaslicer from Appdata, i need to reinstall and log in on bambu lab to download my profiles.

> BTW have you tried disabling anti-virus software?

When i use other profile without remote folder on Appdata, (c:\users\admin\appdata\roaming\orcaslicer\, can open using the same computer without any problems.

Noisyfox, 2025-05-29:
> Yes, enabled and installed. When i delete the folder Orcaslicer from Appdata, i need to reinstall and log in on bambu lab to download my profiles.

Have you tried not installing the network plugin and see if it works?

pcdrivers, 2025-05-29:
> I have my OrcaSlicer (2.3) AppData pointed to another local drive and the startup time is about 90 seconds. So I'd be curious if maybe the network drive adds just enough extra delay to cause a crash.
> 
> Windows has always (going back to at least the XP days) had a bug (or design flaw) where attaching to a network drive that hasn't been used in awhile (e.g. connection may have been slept) blocks other file system IO momentarily until it can connect, to a multitasking user this can appear as a "hang" or stutter (depending the lockup period) and cause havoc for apps trying to access file IO without a retry.
> 
> What happens if you open the network folder in File Explorer (thus establishing a fresh connection) before launching OrcaSlicer?

That is not a problem, because its a company network, and the computer dont have problems with other problems.

> > Yes, enabled and installed. When i delete the folder Orcaslicer from Appdata, i need to reinstall and log in on bambu lab to download my profiles.
> 
> Have you tried not installing the network plugin and see if it works?

I’ve done some testing, and obviously I realized that this prevents me from connecting to the printers via Bambu Lab.

And it seems the exact same issue occurs.

Strangely, on another computer running Windows 10 Pro, I don’t experience the same problem.

Last night I left the computer with the program trying to open and went home. This morning I found the program open, and a Procmon log of over 500 MB, but no errors. It's quite strange — I don’t know how many hours it took to open, but it mysteriously launches quickly when there are no configuration files. It creates the files, and then it seems that accessing the configuration files from the network takes an extremely long time.

I’ve been running network tests this morning, especially after my colleague’s message, and there don't appear to be any network issues. In fact, all desktop and system files load from the network, since these are domain-joined machines. I even tested with the same network cable and switch, on a machine running Windows 10 Pro, and it worked perfectly.

I also checked for packet loss on the network card, and there doesn’t seem to be any issue.

It’s something extremely odd and most likely related to Windows 11 Pro and the remote network folder.

Noisyfox, 2025-05-29:
> Last night I left the computer with the program trying to open and went home. This morning I found the program open,

Do you have the app log for this instance?

pcdrivers, 2025-06-06:
I detected the problem while testing.
If I delete the history of open projects from the configuration file, it opens normally.

Noisyfox, 2025-06-06:
Did your history file also sits on a remote folder?

closed 2025-09-13 by github-actions[bot]