OrcaSlicer/OrcaSlicer #9733: I have two questions, when will the filament exchange system be available in Orca, what is the perimeter overlap parameter?

louro67, 2025-05-27:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I have two questions, when will the filament exchange system be available in Orca, what is the perimeter overlap parameter?

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

I have two questions, when will the filament exchange system be available in Orca, what is the perimeter overlap parameter?

Noisyfox, 2025-05-28:
What is filament exchange system?

louro67, 2025-05-28:
and it should also have a parameter like Simplifly 3D has for overlapping perimeters, as this is important for the last layer to be perfect!

Noisyfox, 2025-05-28:
> AMD sync system how to have bamboo studio!

Do you mean this?
[image]

louro67, 2025-05-28:
yes that's it!