OrcaSlicer/OrcaSlicer #9736: Bambulab X1C and OrcaSlicer 2.3.0 using Bambu Connect fails to load filament from AMS

digitalresistor, 2025-05-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

macOS

### OS Version

MacOS 15.5

### Printer

Bambulab X1C

### How to reproduce

Bambulab X1C firmware: 01.09.00.01 (Latest version)
AMS firmware version: 00.01.06.62
 
3. Create new project in OrcaSlicer for the Bambulab X1C, for example the flow rate test
4. Selected desired filament for the X1C
5. Slice the plate
6. Cmd + G to export G-Code 3MF file
7. Use Bambu Connect to send the G-Code file to the printer

### Actual results

The printer will unload any filament that is loaded from the AMS, and then refuse to load new filament stating that the AMS is not sending any filament.

### Expected results

I expect the print to load the selected filament and start printing

### Project file & Debug log uploads

Anyone can re-create the project in OrcaSlicer by creating a brand new project. This is not specific to any single project, it is an issue with ANY project sliced with OrcaSlicer being sent to the Bambulab X1C.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-05-28:
https://github.com/SoftFever/OrcaSlicer/pull/9517
Please try this

zpriddy, 2025-05-31:
@Noisyfox testing with your PR did not fix the issue related to filament selection. I am guessing that there is a G-Code update that is required to be gingerly lifted from Bambu Studio.

[image]
[image]

digitalresistor, 2025-05-31:
@Noisyfox @zpriddy is my friend who I am helping test this, so I can confirm the same.

AliceTrooper, 2025-05-31:
Here the same.
The Job was send to P1S. Heatbet was heating, nozzle is at target temperatur, but the AMS won't load the filament.
In Orca Slicer the state ist Hotbed heating.

Unable to print some parts

Noisyfox, 2025-05-31:
> [@Noisyfox](https://github.com/Noisyfox) testing with your PR did not fix the issue related to filament selection. I am guessing that there is a G-Code update that is required to be gingerly lifted from Bambu Studio.
> 
> <img alt="Image" width="695" src="https://private-user-images.githubusercontent.com/1858679/449443779-93f09c20-378f-4454-8a3e-e147be7c0436.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg2NzcwNDgsIm5iZiI6MTc0ODY3Njc0OCwicGF0aCI6Ii8xODU4Njc5LzQ0OTQ0Mzc3OS05M2YwOWMyMC0zNzhmLTQ0NTQtOGEzZS1lMTQ3YmU3YzA0MzYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MzFUMDczMjI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZGM0Nzg1N2JmNGVmYWM5NGNmYmVkZTRiNjc0ZjY2Y2M5YjhlZmQwOTIxNTVmMTllZDM3ZTgxMDRhYTVlM2IzOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.JhgaZyuHZI6UoTQ5EP0Uw5fmXYCRHHQLXeVz4Hg5lvE"> <img alt="Image" width="347" src="https://private-user-images.githubusercontent.com/1858679/449443780-6e3e1e00-a97a-442b-83b7-cffd3a4173b1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg2NzcwNDgsIm5iZiI6MTc0ODY3Njc0OCwicGF0aCI6Ii8xODU4Njc5LzQ0OTQ0Mzc4MC02ZTNlMWUwMC1hOTdhLTQ0MmItODNiNy1jZmZkM2E0MTczYjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDUzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA1MzFUMDczMjI4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzQ1NWRhZGM4MmE3MmViMGVjODAwNmNjODJmZjAyNGQ1MTQ4ZmNiZWFkZTY4MWNkZGU2YzQ1ZTZlMGNhMjI5NCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.aq1mu2YPHDq-A9ktWK33ZDQWLam0IJzONpAlCfi5qkg">

Have you disabled legacy network plugin?

AliceTrooper, 2025-05-31:
[image]

I don't have legacy network plugin.

Like I said.
The Job was send to the printer. The bed is warm, and the nozzel to. The Hotend drive like with bambu studio. 
The step, when the printer select the AMS slot and fill the filament don't work with Orca. The printer yust do nothing.

Noisyfox, 2025-06-01:
> [image]
> 
> I don't have legacy network plugin.
> 
> Like I said. The Job was send to the printer. The bed is warm, and the nozzel to. The Hotend drive like with bambu studio. The step, when the printer select the AMS slot and fill the filament don't work with Orca. The printer yust do nothing.

You are using the wrong version. Check https://github.com/SoftFever/OrcaSlicer/pull/9517

digitalresistor, 2025-06-03:
> Have you disabled legacy network plugin?

Yes, the legacy network plugin was disabled. The new plugin was loaded and running.

Noisyfox, 2025-06-03:
Do you have multiple ams connected? Could you take a screenshot of the device page? Thanks

digitalresistor, 2025-06-03:
> Do you have multiple ams connected? Could you take a screenshot of the device page? Thanks

Single AMS. I left my friends place so I can't connect to his printer anymore. I'll upgrade my own printer and test when I am done with the prints I am working on right now.

AliceTrooper, 2025-06-04:
Deactivating the legacy network plugin is the solution!

digitalresistor, 2025-06-05:
Screenshot of the settings:

[image]

Screenshot of the Bambu X1C running firmware: 01.09.00.01
[image]

Noisyfox, 2025-06-05:
I think it might have something to do with the fact that your AMS is recongnized as the second AMS (see the B1,B2,B3,B4 on top of each slot instead of A1A2A3A4).

> using BambuConnect to send the file

Did you export the sliced 3mf using the version from PR https://github.com/SoftFever/OrcaSlicer/pull/9517 as well?

Noisyfox, 2025-06-05:
https://wiki.bambulab.com/en/software/bambu-connect/releases/v1_2_2-beta_2

[image]

Are you using the latest version of BambuConnect?

digitalresistor, 2025-06-05:
[For Delta flowrate_0_PLA_37m3s.gcode.3mf.zip](https://github.com/user-attachments/files/20600725/For.Delta.flowrate_0_PLA_37m3s.gcode.3mf.zip)

As requested, here is the G-Code sliced file.

digitalresistor, 2025-06-05:
@Noisyfox I am not sure why slicing from BambuStudio worked for me then, maybe I made a mistake somewhere, entirely possible? Thank you for the follow-up, but it looks like this can be closed and I will file a bug against Bambu Connect with Bambulab.