OrcaSlicer/OrcaSlicer #9758: Klipper via the “device” tab

ZoskyZosk, 2025-05-30:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Linux Mint 20.2 Cinnamon

### Additional system information

Intel© Core™ i5-2500K CPU @ 3.30GHz × 4
Ram DDR3 24Go
NVIDIA Corporation GK104 [GeForce GTX 660 Ti]

### Printer

Voron Switchwire

### How to reproduce

After installing KAMP on Klipper, if I access Mainsail via the “DEVICE” tab and consult the configuration files in the KAMP directory: Adaptive_meshing.cfg, KAMP_Settings.cfg, Line_Purge.cfg and all the others, they are completely empty, no lines of code appear. 

On the other hand, if I use klipper's Web interface to access Mainsail, everything's fine, and the files mentioned above are filled in correctly. 

The problem is that Orca doesn't take into account certain configurations and additions. 

If you need to modify, correct or test configurations, you'd better use the web interface, otherwise you'll waste a lot of time looking for problems where there are none.

[image]

### Actual results

See explanation above

### Expected results

Access to Mainsail via the “DEVICE” tab should accurately reflect the file system and configuration present in the Mainsail web interface, and take into account live modifications.

### Project file & Debug log uploads

https://github.com/kyleisah/Klipper-Adaptive-Meshing-Purging/tree/main

### Checklist of files to include

- [ ] Log file
- [ ] Project file

C8forT, 2025-07-24:
I have a similar issue.  My "Device" tab presents Mainsail.  Everything seems to work EXCEPT the following:

1.  If I select "G-CODE FILES" from the left menu in the Mainsail interface there are no g-codes listed, even g-codes I uploaded from Orca are not listed.  The Current path is listed as /gcodes.  The actual location is ~/printer_data/gcodes, and my printer.cfg file has:

 [virtual_sdcard]
path: ~/printer_data/gcodes

2.  If I select "MACHINE" in the Mainsail interface there are no configuration files listed.  The root directory is listed as config.  The actual location is ~/printer_data/config.

Everything shows as it should if I open Mainsail using Firefox or Chrome.  Also, in both browsers, the Current path and root directory are indicated the same as indicated in Orca slicer, i.e., "/gcodes" and "config", respectively.

 Thus, the issue appears to be with Orca Slicer.  I have the issue both with version 2.3.0 (Ubuntu) and version 2.3.1 (flatpak).  My primary OS is Linux Mint  22.1, but I also have the same issue with Arch Linux using version 2.3.0 (flatpak).

sebastianha, 2025-09-20:
Same here, when opening Mainsail via OrcaSlicer (flatpak) some parts of the UI are missing. E.g.: console entries & temperature charts. Some parts display fine, some are just missing. In Chrome everything is fine. On Windows also everything is fine when using OrcaSlicer, only Linux Flatpak seems to be broken. (2.3.1-beta)