OrcaSlicer/OrcaSlicer #9767: Synchronous Across-Device Setting Updating

invictus-anic3tus, 2025-05-30:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I own two computers, one I use for working and the other I use for free time and hobby-ing. The hobby computer has Orcaslicer updated to the newest version and all the perfect settings and filament profiles for my printer. But sometimes I need to get a print started when I only have access to the work computer, which has Orcaslicer installed, but all the filament profiles and print settings are either outdated or flat-out wrong. It takes forever to manually go through and fix each and every one, with both computers open, and exporting the settings and importing on the other computer can be a hassle.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

What I propose is a sort of user system. From my experience with web dev, I know that databases and stuff like that can be difficult to implement, but I think it'd be much worth it. All you'd need to add is some sort of simple sign-in page, even just using GitHub or Google. That way, settings can be synchronized across two or more computers and fix issues like my own. It'd really revolutionize the slicer, and, as far as I know, no other slicers have done so yet!

### Describe alternatives you've considered

Instead of a database user system, some sort of setting where, on computer #1, you can link computer #2? I have no idea how this would work. Something with IP addresses maybe?

kens96, 2025-05-31:
My settings synchronise across devices perfectly when longed in to the Bambu cloud.

invictus-anic3tus, 2025-05-31:
What is Bambu Cloud? I've never heard of that and don't see a setting for it.

hh-systems, 2025-05-31:
I would prefer not to use some corporations servers. perhaps a git sync? you could use github or any other git server. I like this idea. allot. I'm just starting to teach my kids 3d printing and having to manualy transfer all the setting from my orca to the computer im setting up for my kids is a little cumbersome.