OrcaSlicer/OrcaSlicer #9786: Merge outer layer on modifier

su-sd, 2025-06-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

If I use a modifier, the slicer creates multiple new seams, where the modifier is placed.
I understand, that the modified parts are sliced separately and the individual parts are merged together
Sometimes it is not possible to move the modifier just inside the object (e.g. if it is has a lot of angles). 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

To improve the quality, it would be nice if the most-outest layer line is not included, so there are no additional seams.

tlhintoq, 2025-06-03:
> If I use a modifier, the slicer creates multiple new seams, where the modifier is placed.

I've never encountered that.  "A modifier" is pretty vague as modifiers can do a wide range of things.

I'd urge you to include a full solution file for the developers: So they can see it happen, then they can move the modifier off the object and see it _not_ happen.

su-sd, 2025-06-03:
What I used: Wall loops 2, the modifier wall loops: 4

Do you want a 3mf file which reproduces the behavior?

JamesHagerman, 2025-06-04:
I believe I ran into this same issue. Please correct me if I'm wrong!

I wanted to print a part that used 4 walls for most of the model but then 10 walls around some vertical holes.

Adding a cylindrical modifier object and setting 10 walls seems to causes the slicer to treat the outer wall of the modifier as separate from the actual object I'm trying to print. This causes unexpected artifacts.

[image]

[image]

[image]

kay0815, 2026-02-10:
would be great to see this feature implemented