OrcaSlicer/OrcaSlicer #9792: Tree supports causing layer defects?!

igl-x2, 2025-06-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11Pro 24H2

### Printer

P1P

### How to reproduce

It seems the generation of Treesupport (style: hybrid) causes random layers to be printed crappy, making the part weak and easy to break.
Take a look at those models, all requiring supports and all failed because of a single improperly printed layer. Height and position of the failure seems to be random. Sometimes the error happens in the model, sometimes in the support branches.
I used different filaments, checked the whole extruder setup, reduced printing speed to no avail.
Not every printed item suffers from this. I had 3 objects on a plate and 1 was ok.
Objects without tree supports print fine 100%.
.3mf files are available, but too big for me to check the gcode manually

Solution so far is:
switching to BambuStudio slicer. BBS delivered two flawless prints in a row. However the visual quality of the parts was clearly behind Orca.

### Actual results

[image]
[image]
[image]
[image]
[image]
[image]

this is a closeup of what the broken layer looks like:

[image]

### Expected results

sometimes, I get a model that prints flawless (Orca):

[image]
[image]

And these 2 are from BBS

[image]

### Project file & Debug log uploads

[PPTower.zip](https://github.com/user-attachments/files/20581467/PPTower.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

tlhintoq, 2025-06-04:
If I saw those photos on any FB post - my first call would be that you had a partial clog that cleared itself with pressure build up.
The seams in your tree support are some massive trenches.  Personally, I think its a settings issue.
The test if this is a genuine slicer issue is if it happens every time at the same place.

TheFlyingBanshee, 2025-06-04:
I have the same issue, but in my case it occurs at the same layer height on every print. 

It seems to be something to do with the extra little bit of wall OrcaSlicer adds to re-inforce the support seem floating in space/being unsupported/being an overhang.

tlhintoq, 2025-06-06:
> *Edit for clarity. Every print of that object.

Do you want to attached the STL and orca 3mf so someone can look at them?

LoghamLogan, 2025-06-16:
Thank you for raising this! I have the exact same issue on a Bambu X1C. I replaced the entire hot end thinking it must be clogging, but thinking more about it, it's always a perfect horizontal line (on the same layer). It's too consistent to be a clog. It only started happening after upgrading to 2.3.0 and I also have the same defects when using the "Default Slim" option for trees as well.
Switching to "Default Grid" now and seeing if it stops happening.

tlhintoq, 2025-06-17:
> it's always a perfect horizontal line (on the same layer). It's too consistent to be a clog. 

Ummm... That's how these printers work: In horizontal layers.  So if you get a clog its going to be perfectly horizontal until enough pressure builds back up in the nozzle to blow the clog... or your zHop lets the clog out... retraction/detraction at the end of the layer gets it loose

LoghamLogan, 2025-09-15:
> Ummm... That's how these printers work: In horizontal layers. So if you get a clog its going to be perfectly horizontal until enough pressure builds back up in the nozzle to blow the clog... or your zHop lets the clog out... retraction/detraction at the end of the layer gets it loose

I can reproduce this by turning on the feature, so maybe I can attach some gcode to get it reproduced?

There's just no chance it's a clog.