OrcaSlicer/OrcaSlicer #9794: STEP file manual import quality control

GuardianArias, 2025-06-04:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

At some point STEP files where nerfed when compared to high quality STL files. the used to both be imported and look indistinguishable from each other but the STEP file would be several gigabytes smaller for large complex parts.
This lower quality translates into actual printed parts; I've been chasing a spiral VFA assuming it wasn't the slicer but upon close inspection it was the slicer.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be great to have the ability to choose the mesh quality of the imported STEP, please let me make my own mistakes. Even if I can simply go into a config file and edit the mesh quality that would be fantastic.

Attached are pictures of an object that easily showcase the VFA I'm seeing on actual printed objects

### Additional context

[image]
[image]

Orel-A, 2025-07-23:
[image]

Problem explained here:
https://www.youtube.com/watch?v=1P0_14A_q60
Summary of the problem: Orca Slicer imports STEP file in low quality, not enough resolution for art models.

What appears to give good enough quality is those values:
Linear Deflection of 0.003 and Angle Deflection of 0.02
Wiki from Bambu: https://wiki.bambulab.com/en/software/bambu-studio/step

Felix14-v2, 2026-09-30:
@GuardianArias quality settings are now available. Could you check the latest version and close the issue if it meets your needs?

RF47, 2026-10-04:
quality settings are now available

closed 2026-10-04 by RF47