OrcaSlicer/OrcaSlicer #9796: Orca Slicer Cloud Prints Blocked with update 01.08.02.00 on P1S (DON'T UPDATE)

1ubuntuuser, 2025-06-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Linux

### OS Version

Debian

### Printer

Bambu Labs p1s

### How to reproduce

Update to latest firmware on p1s (01.08.02.00)
Try printing from the slicer.

### Actual results

Fails silently when the screen is switched to the device page.  Release notes do say this is due to security. 

### Expected results

Printing with cloud

### Project file & Debug log uploads

Not required

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Knew it was coming but didn't expect it this soon.

I've marked it as a bug as the UI should tell you why this has happened and advice you go to the bambu handy app to downgrade the software.

IntBoat, 2025-06-04:
A1M / A1 is experiencing the same issue; Developer mode and LAN mode must be enabled in the printer to use third-party software.

1ubuntuuser, 2025-06-04:
> Not a bug since it is entirely a Bambu Lab issue which Orca has no control over. Frankly, it's not really OrcaSlicer's responsibility to inform users that their printer is not compatible due to the firmware, or anything really, and officially telling users to downgrade in items liability. I think this should be entirely on the user to 

Apps should work with users in mind. I get what you are saying, but the Orca Slicer auto-installs the Babu Labs network plugin, which does not work. For new users who unwittingly try running OrcaSclicer, it will just fail to connect with no indication of why.

@IntBoat Did you have any good instructions on how to set this up?  Might be helpful for others.

IntBoat, 2025-06-04:
@1ubuntuuser 

Simply enable LAN Mode in the printer settings, and a Developer Mode option will appear on the interface.
Turn on Developer Mode, which will disable the 'security certification' recently added by Bambu Lab that prevents third-party software from connecting.

Finally, in the Device interface of the Orca software, use 'Bind with Access Code' to use Orca normally.
However, the trade-off is that the printer will no longer be controllable or printable through Bambu Handy, and you won't be able to use Handy to view the camera.

Additionally, the official website still allows downloading the previous version of the firmware, but in the long run, it won't be possible to avoid updates forever.

[image]

[image]

[image]

[image]

MxBrnr, 2025-06-07:
> Orca Slicer auto-installs the Babu Labs network plugin

As a matter of fact, it doesn't unless you tell it to during initial software setup, and even that can be retracted at any time via the settings in Preferences.

boysie123, 2025-06-08:
just roll back to the previous firmware you lose nothing and gain everything. check the release notes of the firmware